bancika/diy-layout-creator #407: curve traces render 10% thicker.

kroma-1, 2019-01-18:
I opened a project created in V 46.1, and curve traces appear 10% (aprox.) thicker, although they are 1mm. 
I´m attaching 2 screenshots. Pay attention to the second one, the bottom black line is a straight line, 2mm., and the red on top is a "rectified" curve, same width.

I know this is a small issue, but its visually annoying (OCD speaking)

[image]
[image]

bancika, 2019-01-18:
not sure what the issue is since your screenshot shows two different layouts :) in which version do you find the traces to be ticker? Are those curved traces? Your second screenshot has a straight trace selected. Not sure where I'm supposed to look at :)

bancika, 2019-01-18:
maybe I understood what you were saying (although I'm still not sure where to look at)...curved traces are rendered 1px wider than straight traces of the same thickness. If that's the case, I'll fix for 3.50

closed 2019-01-18 by bancika

kroma-1, 2019-01-18:
Those are 2 sections of the same layout. The red one in the second capture is not a selected trace, its a curve but straightened and superimposed on the black one.
Also, I marked where a curves join straight traces.
[image]

kroma-1, 2019-01-18:
I´d say about 2px wider... ;)
Thanks.

bancika, 2019-01-18:
gotcha, yeah, I think I fixed it. Thanks.