bancika/diy-layout-creator #412: Selecting a jumper sets it colour to red

timsavage, 2019-01-24:
Version: 50.0
OS: GNU/Linux - Ubuntu 18.04 64bit
JRE: Java HotSpot(TM) 64-Bit Server VM (build 9.0.4+11, mixed mode)
Log: [log-2019-01-24 06.17.58.txt](https://github.com/bancika/diy-layout-creator/files/2791075/log-2019-01-24.06.17.58.txt)

When selecting a jumper it goes red as expected, however, after selecting another component the previous colour is not restored. Even after selecting an alternate colour (via edit dialogue). This doesn't seem to affect hookup wires or other components.

Steps to reproduce
- Draw a jumper wire
- Select the jumper

bancika, 2019-01-24:
hmm, I just tried and it worked fine, although under windows 10.
Does it work ok with earlier versions, like 3.45?

timsavage, 2019-01-24:
After doing a bit more testing I've found that this also affects resisters when they are standing up.

Tested with 3.48, this version does not exhibit the same issue.

EL34xyz, 2019-01-25:
Ok, I see now that selecting the jumper changes it to red permanently as was reported
I was thinking that sending it backward was the culprit
This is in 50.0 on Windows 7 pro

EL34xyz, 2019-01-25:
I am not seeing a problem with resistors in 50.0  as was reported. 
Can you tell me the exact component you used?

EL34xyz, 2019-01-25:
Ok, I see what timsavage was saying about resistors
I think this is an issue of changing the angle of a component
I just saw the leads of the resistor on the left change to red after changing the angle or orientation

[image]

EL34xyz, 2019-01-25:
If you open the edit selection menu, you can change the Red leads back to the original color. 
FF0000 is the color that the leads were changed to
CCCCCC is the original lead colors that I use

[image]

bancika, 2019-01-28:
fixed for 3.51

closed 2019-01-28 by bancika