bancika/diy-layout-creator #413: Feature request - Export and import Variants and Building blocks

EL34xyz, 2019-01-24:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

EL34xyz, 2019-01-24:
It would be nice to be able to export all my variants and Building blocks so that others can import them.
I have about 100 variants and 20 or so building blocks that other people would like to have.

bancika, 2019-02-02:
Maybe we can implement this as a "premium" feature? Email me if interested.

EL34xyz, 2019-02-07:
Thanks, will do

bancika, 2019-02-11:
released in 3.53

closed 2019-02-11 by bancika

iiimonfiiire, 2019-03-25:
This is not working in v3.56 or v3.57 on MacOS Mojave 10.14.3. 

Reproduction steps:
1. Go to File > Integration > Import Variants...
2. Select variants.xml from some directory (see attached)

Result:
An error message reads: "Could not import variants: java.lang.ClassCastException: java.util.HashMap cannot be cast to org.diylc.common.VariantPackage"

[variants.xml.zip](https://github.com/bancika/diy-layout-creator/files/3001919/variants.xml.zip)

bancika, 2019-03-25:
what is the earliest version that works for you? I'm looking at the history
and it seems like your variants file was made with a pre-release version.
Variants are officially introduced in 3.53 and they use VariantPackage that
is missing in your file.

iiimonfiiire, 2019-03-25:
My variants were working up to v3.51, but I have not used this new feature introduced in 3.53. Are we talking about variants.xml that is found in the Package Contents?

EL34xyz, 2019-03-25:
Import and Export variants and building blocks is what was new in 3.53
You could not have done that before 3.53

Not sure what you are asking when you say variants were working in 3.51
Creating variants has never stopped working

bancika, 2019-03-25:
Variants that are part of the package are not meant to be imported by you,
as they are automatically imported by the app. Import/export is for users
to share their variants between them.