bancika/diy-layout-creator #414: Ohm not selectable in resistor drop down menu

EL34xyz, 2019-01-25:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

EL34xyz, 2019-01-25:
50.0 - windows 7 pro

There is some sort of funky characters in the value drop down box for resistors
Ohm symbol is not there

[image]

bancika, 2019-01-28:
fixed for 3.51

closed 2019-01-28 by bancika