bancika/diy-layout-creator #415: Text is not copied to Windows clipboard

EL34xyz, 2019-01-25:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

EL34xyz, 2019-01-25:
50.0 - Windows 7 pro
I was trying to copy a text name from one component to another one in the edit selection menu
It appears as though text is not copied to the windows clipboard in DIYLC

Opening notepad and trying to paste copied text labels from DIYLC results in nothing being pasted in notepad.
Even though I did have something in the windows clipboard, it has apparently been erased and now  the windows clipboard is blank.
Copying text from notepad and then pasting  to a component name works fine.

So DIYLC is getting the text from the windows clipboard and pasting it just fine
DIYLC is not copying to the windows clipboard
DIYLC is erasing the windows clipboard after copying something in DIYLC

bancika, 2019-01-28:
just tried and it worked for me. Anyone else?

EL34xyz, 2019-01-28:
Let me know if this works for you
Open notepad, type a word, copy that word
In DIYLC, copy something
Go back to notepad and paste
What is pasted in notepad?

bancika, 2019-01-28:
for me it works as expected. If I go to Edit Selection and copy component name or value or anything, that's what I get pasted in notepad. But if I go and select a component, copy the whole component and try to paste, nothing gets pasted because the content of the clipboard is not a text, but a whole component.

EL34xyz, 2019-01-28:
Weird, it is working just fine for me now
I have no idea why :)