bancika/diy-layout-creator #416: OSX sidebar/tree bug

MacFly77, 2019-01-25:
Great to have V3.50 here!

There is a thing that still bugs me. 
On OSX 10.11.6, Java 8 201 the sidebar/searchable tree graphics are incorrect when scrolling.
See the attached image. Scrolling slowly helps to refresh....but still.

[image]

FentonBresler, 2019-03-13:
Still an issue here in 3.56.0

bancika, 2020-03-27:
Does it still happen with version 4 and upgraded Java? I can't reproduce it on Catalina.

bancika, 2020-11-19:
the tree was recently rewritten from scratch. Closing for now, please reopen if still an issue.

closed 2020-11-19 by bancika