bancika/diy-layout-creator #418: Possible Memory Leak

robrobinette, 2019-01-27:
My computer has slowed dramatically the last two mornings. Yesterday the computer fully locked up requiring a reset. I have an Intel i7 4 core and the CPU usage was hitting 90%. This morning when my computer slowed dramatically (jumpy mouse movements) I started closing programs and couldn't close one instance of DIYLC (I had 3 instances with complex
 amp layouts and schematics running overnight). I had to kill javaw.exe using Resource Monitor and then everything went back to normal. javaw.exe was only using about 1 to 3% of CPU but Windows System and Memory Compression were using 40 to 80%. I'm thinking DIYLC or java itself has a serious memory leak. I'm using up to date Windows 10 x64 Pro.
[log-2019-01-26 09.45.36.txt](https://github.com/bancika/diy-layout-creator/files/2800189/log-2019-01-26.09.45.36.txt)

robrobinette, 2019-01-28:
I reinstalled 32 bit java but I had the same problem after leaving 3 instances of DIYLC open overnight. I'll try to install 64 bit java and see if that helps.

bancika, 2019-01-28:
It probably won't make a difference...what does the memory bar show in DIYLC? Go over the icon in the bottom right, it will show memory usage stats of that particular instance. If you click on it, it will collect unused memory.

EL34xyz, 2019-02-07:
How long did you leave DIYLC running Rob?

I could do a test
Restart my windows 7 desktop and start with a fresh copy of DIYLC runnig
Take a screen shot of memory in task manager and the icon in DIYLC
Let it run all night and see what happens in the morning

I would probably have to load a file also

bancika, 2019-02-07:
I made a change for 3.53 to manage drawing very differently than before. It should be faster to render and also faster to release unused memory. I would wait for the new version before any testing.

I also assume it has to do with layout physical size, Rob's files are very big and take a lot of memory. That should be vastly improved in 3.53.

Cheers

EL34xyz, 2019-02-07:
Ok, sounds good
Maybe Rob can test the new version with the exact same file and see what happens

robrobinette, 2019-02-07:
Yes, I will upgrade to 3.53 as soon as it's available. I updated to 64 bit java and disabled the 32 bit version but I had the same problem. I checked the memory usage in DIYLC and it wasn't excessive. It seems to take about 4 to 6 hours for the problem to come up. Like I said in the first post, Windows System and Memory compression services were using most of the CPU.

robrobinette, 2019-02-13:
I left DIYLC 3.53 running overnight and as soon as I "unminimized" DIYLC my computer slowed to a crawl. I opened Task Manager and "Windows System" and java were consuming over 90% of the cpu. I killed java and the computer is back to normal. I'm running up-to-date Win 10 x64 Pro and 64 bit up-to-date java.

bancika, 2019-02-13:
Thanks. Do you have any file in particular open at that time? Maybe you could share it so I could try doing the same on my machine.

robrobinette, 2019-02-13:
Here's the file I had open overnight: [https://robrobinette.com/images/Guitar/AB763_Models/RR763_BLACKVIBE/RR763_BLACKVIBE_Layout.diy](url)

bancika, 2019-02-13:
Thanks, I'll leave it overnight.

bancika, 2019-02-14:
Interesting, nothing happened on my win 10 x64 machine. It still runs smoothly with no increase of memory or CPU usage overnight. My java is 1.8.0_161

robrobinette, 2019-02-14:
I tried to attach my config.xml file and got a "We don't support that file type." error.

bancika, 2019-02-14:
it's possible only zipped.

robrobinette, 2019-02-14:
Here you go:
[config.zip](https://github.com/bancika/diy-layout-creator/files/2865706/config.zip)

bancika, 2019-02-14:
Thanks! I don't see much in the config, it's pretty much stock, so it's safe to say that config is not to blame. I'll try to update java and see if it changes anything. Other than that, PCs are very close. Mine is also i7 with win10.

bancika, 2019-02-18:
Left it over weekend with the latest java (1.8.0._201) and it's still running perfectly :/

robrobinette, 2019-02-18:
It must be something unique to my computer. Feel free to close this issue.

closed 2020-12-13 by bancika