bancika/diy-layout-creator #421: Ohm symbol in v3.52.0

cctsim, 2019-02-03:
In v3.52.0, if I create a new project, the Ohm symbol is displayed as a square in resistors. 

However, if I open a project created with an older version (like v3.49) the Ohm symbol is displayed correctly and newly created resistors look also fine.

[log-2019-02-03 02.11.53.txt](https://github.com/bancika/diy-layout-creator/files/2825589/log-2019-02-03.02.11.53.txt)

bancika, 2019-02-04:
I think it has to do with the font. New default font "Square" doesn't show ohm symbol on some systems. Can you try some other font, like Lucida Console? Or Tahoma, the previous default font.

cctsim, 2019-02-04:
That's it, thanks. I changed the font to Ubuntu Mono and it works fine. Is there a way (instruction) to add this choise to the config file so that the change is permanent ?

bancika, 2019-02-04:
yeah, just select the checkbox to the right side of the font to make it a
global default. It works for all properties of project or components. Handy
:)

closed 2019-02-05 by bancika