bancika/diy-layout-creator #425: Circle/wiring functionality test | Most important (next level)

mspeace, 2019-02-05:
This program is for illustrative purposes only really but it does it very well (better than any other program we tried for producing diagrams for guitars)

but!

now I’ll talk only about electric instruments
in my mind the idea of this program born as a complete builder and tester of wiring 
(will show you is it right, and will it work or not (maybe with tips how to fix bad wiring)

If we dream a little more, you could even set the desired sound and pickups that you have
then the program would pick up the necessary wiring by itself (need database)
maybe also with the displaying of all the existing forces (currents) that goes through it

in the most insane way to collect the real sound of different pickups and make an ultimate thing!!!
but it will took a time

i will be happy to see this in your app
will be happy to help you with it

and then you should jump into an mobile app

a lot of people will appreciate it, it will save a lot of time/nerves/whatever

bancika, 2019-02-05:
thanks.

I have thought about similar ideas too. It's a very ambitious project for sure as there are many variables to consider. When I have free time I'll think about it more and see if there's a feasible road-map to get at least some of that done.

mspeace, 2019-02-05:
yea i wrote it in spontaneous way
but about " to get at least some of that done"
i think firstly - wiring builder (tester)
just because im not so good at it, im tired and bored, got burn on finger
(i'm ok :)
bored to search wirings on internet, and after i guess, 6 attempt, i left
my guitar only with one gibson humb. (because second is very unique)
thats it.
gonna fix it

bancika, 2019-02-06:
Before I invest several weeks of my time into this I need to know that it will benefit a bigger group of users. I rarely hear from folks using DIYLC for guitar wirings, so not sure how many would end up using this feature.

Just because you are not so good at it doesn't seem to be a strong reason :)

EL34xyz, 2019-02-07:
I never used the guitar stuff in DIYLC.

mspeace, 2019-02-07:
uh
right

bancika, 2019-02-07:
The math geek inside me wants to do it, though :)

bancika, 2019-02-26:
Following up, I released a beta version with some of the features I described, so people can poke around and get a feel of it. You can find the latest release here https://github.com/bancika/diy-layout-creator/releases
Also, I have attached to this email one of the files I used to test the app. It has few complexities (like treble bleed, push-pull split, hum-canceling middle split position, etc), so it's a good test case. Feel free to try any other combination of switches, pots and pickups. All switch types should work, but I haven't done bass pickups yet, they are still for display only. Humbuckers and single coils work. Single coils have "Polarity" parameter that can be used to set it to North/South for true single coils or Humbucking for double coils. The only requirement is to have one mono jack added to the drawing and have it properly connected. Pickups now have two or 4 terminals depending on type that need to be wired to switches and pots as they would be in real life. Pickup coils are marked N and S for North and South.

The new features can be accessed from the "File->Analyze" menu and here's the sample output for my travelcaster2 file. It's easy to see which coils will be active, how they are wired and what external components will affect the sound, like the Rbleed + Cbleed network that's in parallel with one half of the volume pot, clearly signifying that it's a treble bleed circuit.

Give it a try and let me know how it works for you.

[travelcaster 2.zip](https://github.com/bancika/diy-layout-creator/files/2904728/travelcaster.2.zip)

dorrisant, 2019-04-20:
Anything to show connectivity through a series of parts would be wonderful. At minimum, picking the 1st part in the chain and clicking a button. Then have all directly connected components, wires, lugs, eyelets, etc. would be highlighted, as if selected. This would be wonderful for PCB design as well as any layout. This feature seems like it should be a derivative of some of the above functionality. Should I repost as a separate issue?

bancika, 2019-04-21:
This is already supported through expand selection functionality.

EL34xyz, 2019-04-21:
Config menu - Highlight Connected Areas
Cursor turns to a cross
Click on a connection
All connections in that chain turn bright green
As shown in my screen shot

[image]

TomKamphuys, 2019-08-17:
This is a great step! I've been tinkering (offline) with circuit analysis myself. I 'parsed' the diylc file to generate a netlist for two functionalities I posted about on another forum:

1) Comparing of schematic and layout (https://www.tdpri.com/threads/software-to-compare-layout-and-schematic-in-diylc.888412/)
2) Frequency response  (https://www.tdpri.com/threads/software-to-calculate-bode-plot-based-on-diylc-schematic-layout.917170/)

I think it will be a small step to natively support both use cases and at least I will be using it quite often :-)

closed 2020-03-27 by bancika

mspeace, 2020-03-27:
im lil lazy to read the thread, but whats new? :)

bancika, 2020-03-27:
nothing new. I closed it because I did what I planned to do for stage 1.

mspeace, 2020-03-27:
sorry, hadn't seen 26 feb 2019, and i see that there is gen4 released :)
should i try the travelcaster 2.zip now? :)

bancika, 2020-03-27:
what about it? I forgot