bancika/diy-layout-creator #426: DIYLC PCB Text Lettering Does Not Show Up When Trace Is Exported.

dorrisant, 2019-02-05:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

dorrisant, 2019-02-05:
When using the PCB Text that should show as mirror image, the text does not show up when the trace is exported. So if I label my PCB with my logo and revision number, it gets stripped off when you export the trace as a pdf.

bancika, 2019-02-05:
did you use the latest version 3.52? I fixed the issue there

dorrisant, 2019-02-06:
That last update covered it.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

bancika, 2019-02-06:
Cool, closing

closed 2019-02-06 by bancika