bancika/diy-layout-creator #427: 1/4 jack Close circuit

stinga11, 2019-02-08:
Hi, I wanted to know if it is possible that you add the 1/4 Jack closed circuit. It is used a lot in amplifiers layout. In your wonderful program is not available. Thanks

bancika, 2019-02-08:
not sure I follow, do you have a photo?

stinga11, 2019-02-08:
In the program you have the 1/4 Jack open circuit that has two pin. But in the amps we use the 1/4 Jack closed 
[image]
circuit that has three pin for the speaker output and the guitar input. Here I leave the photo

stinga11, 2019-02-08:
switchcraft 12a

bancika, 2019-02-08:
but you already have that as "Open Jack"
[image]

stinga11, 2019-02-08:
its not the same thing

Stereo(Tip,Ring,Sleeve)
[image]
Mono Closed circuit(Tip,Tip Shunt,Sleeve)
[image]

bancika, 2019-02-08:
I see what you mean now

closed 2019-02-08 by bancika