bancika/diy-layout-creator #429: 10 versions of DIYLC, no recent projects show up, no building blocks

Swampertech, 2019-02-10:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

Swampertech, 2019-02-10:
Bancika, I went to the files on my Mac I was just working on, and added the .diy extension to their names and I could bring up the file in V. 53. But, I still have no building blocks.

closed 2019-02-10 by Swampertech

bancika, 2019-02-10:
It's not clear what the issue is and which steps lead to the issue. Please
elaborate

Swampertech, 2019-02-10:
I went back to V. 52, and tried to save Doug’s templates as variants, but they don’t save or show up in my building blocks or schematics.