bancika/diy-layout-creator #430: Open 1/4" Error after update and trying some ____diy files.

TLBH, 2019-02-11:
Hey bancika,
Just installed latest release..
Love the additions from EL34world (hats off to you and Mr.Hoffman) that you've integrated into this release.
Unfortunately, seems to come with a (albeit small) price.
Hate to be a PITA, but I was checking out one of Doug's layouts (the Re-Vibe) and upon trying to add/initiate a open 1/4" jack I got a: 

> Error
> Could not set component type slot.
> Check log for details.

So I closed the file and tried starting 3 new projects and got the same result just to make sure it wasn't related to the Re-Vibe file.
I use the open 1/4"  because that's what I have as stock (from years of dumpster diving/salvage).
Although not as svelte as in layouts, I use 'em out of habit/visualization. 
Definitely not mission critical, but thought I'd give you a heads up nonetheless.
Keep on keeping on man!
-tlbh

Project #3 log:
[log-2019-02-11 10.33.22.txt](https://github.com/bancika/diy-layout-creator/files/2851997/log-2019-02-11.10.33.22.txt)

Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

bancika, 2019-02-11:
Seems like you have some default value set for one of the Jack's properties. Can you please attach your config.xml file from the user folder. I'll find which one it is.
Cheers

TLBH, 2019-02-11:
Didn't think of that.
Saw all the "com.thoughtworks.xstream.converters.ConversionException"...etc... errors and assumed a java incompatibility (not a first here). 
[config.zip](https://github.com/bancika/diy-layout-creator/files/2853358/config.zip)

bancika, 2019-02-12:
Yeah, this is the problem. In the last version I introduced switched type for open jacks, so I had to move a new enum not to affect other jacks...and it caused this issue. You can remove these 4 lines and it should work. I'll make the process more tight for the next version so it doesn't fail in cases like this. It can just forget about the default and move on.

[image]

closed 2019-02-12 by bancika

TLBH, 2019-02-12:
Unfortunately that didn't resolve it.
Like I said, not mission critical.
Snipped Log:
[ErrorLog.txt](https://github.com/bancika/diy-layout-creator/files/2857159/ErrorLog.txt)

bancika, 2019-02-12:
Hm, it works for me with your config once I removed those lines, please try this config

[config.zip](https://github.com/bancika/diy-layout-creator/files/2857925/config.zip)

TLBH, 2019-02-13:
That did it!
Above and beyond my friend.
Thank You

bancika, 2019-02-13:
you are welcome.