bancika/diy-layout-creator #432: Feature Request: diylc as PortableApps.com package

PH0MB, 2019-02-12:
It's just an idea. 

I have several devices where I'd like to run diylc but running installers are not always convenient or allowed (@work) so it would be nice to have it installed on my PortableApps stick. Since the zip file seems to work OK  suspect it could be possible to have it ported into the PortableApps package and have a dependency for portable java in  the configuration. 

I'll have a try if I can figure out how that works and try to make  a package from the zip file.

bancika, 2019-02-12:
One of the packages I deploy with every release is a zip package that is platform independent and portable. It only writes config to user folder to prevent issues with permissions and write protection.

PH0MB, 2019-02-12:
that's what I'm gonna try to put that into a Portable Apps package. There is some housekeeping needed to integrate it into the Packaging system

bancika, 2019-02-12:
but why going through all that pain when you can just unzip the ZIP file? :)

PH0MB, 2019-02-12:
PortableApps is a pretty nifty toolset and application launcher to publish applications and install all appropriate dependencies without much hassle and without local admin rights and without taking care of installing java runtimes everywhere.  I used the zip before in conjunction with openjdk but it takes some editing in batch files to get it running. I just thought, wouldn't it be nice if it was available as a portableapps package. But I can understand your reservations :-)

ilgrank, 2020-04-12:
I second that proposal. I also would love to use DIYLC wherever I go, but I can't simply because more and more java is not installed where I go.
Having a truly portable version would be great!
Thanks :)

bancika, 2020-04-12:
Anyone knows how to package an app as portable? I'm open to it if someone
can help with that.

PH0MB, 2020-04-13:
I started to make an attempt to create a portable app tree.
The java part is the most difficult thing, because the diylc wrapper needs to find the java runtime somewhere and the launch4j wrapper seems to have some problems with that.
this is also discussed at the  portableapps site documentation 

[https://portableapps.com/manuals/PortableApps.comLauncher/topics/java.html](https://portableapps.com/manuals/PortableApps.comLauncher/topics/java.html)

bancika, 2020-04-13:
No need to use launch4j wrapper, we can start the jar with few arguments
that are specified in l4j config.

PH0MB, 2020-04-13:
OK I'll try that route, I made cmd wrappers before for other java applications so that should work

ilgrank, 2020-04-14:
Thanks for the quick reply guys 👍

PH0MB, 2020-04-20:
I finally have some success in creating a working PortableApps DIYLC envinironment and generating a installer. Due to java path issues diylc.exe does not work so i took the jar instead and javaw.exe
I tried to use OpenJDKJRE64 java tree instead of jPortable (Oracle Licencing) but that does not seem to be recognized as a Java runtime by the PortableApp launcher. Renameing the CommonFiles\OpenJDKJRE to CommonFiles\Java did the trick. I dropped a question in the PortableApps forum to get a clue how this java search works under the hood. I tried reading the delphi source but could not figure out. More because I'm not very familiar with Delphi ;-(

bancika, 2020-04-20:
Great! Let me know when we can integrate it into the official deployment. I have downloaded portableApps, so I guess I would just need to run it against your app descriptor file?

PH0MB, 2020-04-20:
Yup. I carefully followed the instructions of the [https://downloads.sourceforge.net/portableapps/PortableApps.com_Application_Template_3.5.2.zip](https://downloads.sourceforge.net/portableapps/PortableApps.com_Application_Template_3.5.2.zip) and that worked . 
did some resizing of the supplied PNG  and created a .ICO file and 
[DiylcPortable.ini.TXT](https://github.com/bancika/diy-layout-creator/files/4502531/DiylcPortable.ini.TXT)

Is still barfs a bit in generating the .paf.exe installer with the underlying InnoSetup generator
Something about language settings. but It seems to work. I could attach the tree without all de binaries here in a zip if you want?

bancika, 2020-04-20:
I have png and ico files in various sizes, see below

[icons.zip](https://github.com/bancika/diy-layout-creator/files/4502635/icons.zip)

Btw, have a question. DIYLC stores config, logs and file backups in user's directory. In windows it's c:\users\<name>\.diylc

how does that work with portable apps?

PH0MB, 2020-04-20:
Yup, that still works like that, but there seems to be a way to capture that in the Portable App Root tree
There is a wat to capture that as described in [https://portableapps.com/development/portableapps.com_format#hostpc](https://portableapps.com/development/portableapps.com_format#hostpc) and it involved some wizardry in the config files to do that. I have not implemented that for now.
This is what I made, I removed the diylc zip stuff to reduce noise. 

[DiylcPortable.zip](https://github.com/bancika/diy-layout-creator/files/4502703/DiylcPortable.zip)

ilgrank, 2020-04-20:
Hi
Tried it, but still complains the it is lacking Java

bancika, 2020-04-20:
The package seems to be missing all the necessary jars. It should be at
least 6MB large

PH0MB, 2020-04-20:
Sorry I did not make it clear enough  that I removed that to reduce bandwith.
Anyway. I added the .paf.exe.zip that should install it. 

[DiylcPortable_4.4.0_English.paf.exe.zip](https://github.com/bancika/diy-layout-creator/files/4503759/DiylcPortable_4.4.0_English.paf.exe.zip)

You need to install jPortable to run diylc or install OpenJDKJRE64 and rename commonfiles\OpenJDKJRE folder into commonfiles\Java

I've been toying with the launcher DiylcPortable.ini to keep track of the %USERPROFILE%\.diylc stuff but I still don't quite understand the mechanism for copying the config.xml by the PortableApp launcher.

PH0MB, 2020-04-20:
It seems that defining a ini entry like this in the DiylcPortable\App\Appinfo\Launcher\DiylcPortable.ini [FilesMove]

settings\config.xml=%USERPROFILE%\.diylc
seems to keep track of the configuration file issue
I made a change for hardware acceleration and after exiting the program it copies it back into the portableapp data\settings\config.xml

It can be done with whole folders to an d it makes a backup for files and folder already present there.

bancika, 2020-12-13:
Hi,
can you please help me with making the portable build part of the official deployment? What did you do to make that package? How can I automate it?
Thanks!

PH0MB, 2021-01-12:
Hi there again Bane, Sorry to chime in a bit late here. Busy Job and Household. :-)
Happy and Healthy 2021 for you and your family.

I dig into the https://portableapps.com/development page and did some trial and error attempts, but the ZIP I posted earlier is the thing I did. I did not change it any more after that.

Basically  preparing a Portable App shell comes down copy a template, edit a bunch of configuration files and drop the content of the ZIP into the Portable App folder. In my case C:\Portable\PortableApps\DiylcPortable\App\Diylc

I just downloaded the latest one and unzipped it just there and that's it

running the app install creator you can create a diylc<version>-portable.paf.exe installer to install it into the PortableApps start menu. 
It creates an assembly of the zip distribution and the set of needed configuration file and startup shell stuff. 
You can then the whole Portable Application trough the PortableApps Apps/Install a new App(paf.exe)
This is what it looks like

[image]

closed 2021-01-12 by PH0MB