bancika/diy-layout-creator #433: Cannot Launch Java Application

DVAMPS, 2019-02-14:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

DVAMPS, 2019-02-14:
Getting the following error message when I try to launch v53.0:

"Uncaught exception in main method: java.lang.NullPointerException"

I just updated OSX to 10.14.3 and Java 8 201 but issue remains.

Any help you can provide is much appreciated.

tumukei, 2019-02-15:
Looks like a missing file?

```10:35:14,687  INFO [main] ConfigurationManager - Initializing configuration
10:35:14,687 ERROR [main] ConfigurationManager - Could not initialize configuration
java.io.FileNotFoundException: config.xml (M├ñ├ñritetty├ñ tiedostoa ei l├Âydy)
        at java.io.FileInputStream.open0(Native Method)
        at java.io.FileInputStream.open(Unknown Source)
        at java.io.FileInputStream.<init>(Unknown Source)
        at org.diylc.appframework.miscutils.ConfigurationManager.initializeConfiguration(ConfigurationManager.java:86)
        at org.diylc.appframework.miscutils.ConfigurationManager.<init>(ConfigurationManager.java:59)
        at org.diylc.appframework.miscutils.ConfigurationManager.getInstance(ConfigurationManager.java:50)
        at org.diylc.swing.gui.DialogFactory.initialize(Unknown Source)
        at org.diylc.swing.gui.MainFrame.<init>(Unknown Source)
        at org.diylc.DIYLCStarter.main(Unknown Source)
Exception in thread "main" java.lang.NullPointerException
        at org.diylc.appframework.miscutils.ConfigurationManager.copyFileUsingStream(ConfigurationManager.java:118)
        at org.diylc.appframework.miscutils.ConfigurationManager.initializeConfiguration(ConfigurationManager.java:98)
        at org.diylc.appframework.miscutils.ConfigurationManager.<init>(ConfigurationManager.java:59)
        at org.diylc.appframework.miscutils.ConfigurationManager.getInstance(ConfigurationManager.java:50)
        at org.diylc.swing.gui.DialogFactory.initialize(Unknown Source)
        at org.diylc.swing.gui.MainFrame.<init>(Unknown Source)
        at org.diylc.DIYLCStarter.main(Unknown Source)
```

bancika, 2019-02-15:
does the previous version work? Trying to establish when it got broken

schumar, 2019-02-15:
Had the same problem. Tried if 3.50.0 works (it does), and as this old version creates a `~/.diylc/config.xml`, I can now use 3.53.0
So if you want to reproduce, move away your `~/.diylc/`, and then start 3.53.0

bancika, 2019-02-15:
Thanks for the confirmation, I now see the issue. Will include the fix for the next version

bancika, 2019-02-15:
can you please try this build (with moving the config file first)?

[diylc-3.53.0-osx.zip](https://github.com/bancika/diy-layout-creator/files/2869110/diylc-3.53.0-osx.zip)

schumar, 2019-02-15:
I'm on Linux, but managed to start it anyways. I now get a nice dialog telling me that there was an error reading the config file, and that a backup file was created ("with a '~' at the end") -- it wasn't, though.
So it starts now, it creates a `config.xml`, but the message in the dialog is confusing and wrong.

bancika, 2019-02-15:
It looks like under linux it won't recognize there's no config file and detects an error reading it. Under windows it's fine, it creates a file with first use without any warnings.

ICE8, 2019-02-16:
I am using Linux and could not run 3.53.0 until I ran 3.52.0.

bancika, 2019-02-16:
I think I found the issue, will be fixed in 3.54

closed 2019-02-16 by bancika