bancika/diy-layout-creator #434: java memory error

ryand59, 2019-02-20:
Howdy, 

I'm getting a Java/memory error when I try to open an existing drawing.  I'm running DIYLC 3.53.0 under windows 10 (64 bit).

I've attached a screenshot of the error. I can't the current .diylc/logs -- there is a c:/Users/<username>/.diylc, but there is no logs folder or file in that location.  Somewhere else in windows 10 maybe?

I'm pretty new to the tool, but I did not see anyone having exactly this same error already.  As far as I know the drawing files are fine.  It doesn't matter which one I try to open, I always get the same error.

Thanks in advance for any help, 

Ryan D.
ryand1959@gmail.com
[image]

ryand59, 2019-02-20:
OK, 

Sorry but it looks like there are only two drawing files that cause this error when opening.  I tried creating a new drawing and I can save it and open it again without issue.  I also am able to open other drawing files I've created without trouble.  Both the drawing files that are giving me the error are pretty small and can be recreated without much fuss, so this is not a big problem for me.

EL34xyz, 2019-02-22:
Open the problem drawings on an older version of DIYLC
Just download a .zip file and open the .exe in that zip

Did you use one of the older multi section can caps in those drawings?
Those caps will create an error when trying to open the file in the latest DIYLC

bancika, 2019-02-22:
It's probably not that, @EL34xyz . It ran out of memory.

@ryand59, can you please upload the files that give you errors, I'll debug them

ryand59, 2019-02-22:
Thanks guys.  

I don't think I had any caps on these drawings.  They are of humbucker pickups and only the pickups.  I have moved on and recreated the drawings, but here is one of the two files that give me the error.   For some reason, they are very large and even zipped they exceed the upload limit.  

So I created a .png of one of the drawings and I looked at -- I did insert .jpeg images in both these drawings.  Maybe the inserted jpeg's are causing the error and/or bloated size?  I don't recall any other drawings where I did that...

I appreciate your assistance!

Ryan D.

[Original G&B Bridge Pickup.zip](https://github.com/bancika/diy-layout-creator/files/2895007/Original.G.B.Bridge.Pickup.zip)

closed 2019-02-22 by ryand59

bancika, 2019-02-22:
Yes, that's it. DIYLC is not really optimized for handling bitmaps and may consume more memory than needed. Also, files are huge because it uses inefficient way of serializing the data. Your image file is pretty large and takes a ton of memory to handle ( 4160x3120). You can increase DIYLC memory limit to 2GB and then it will work. Just edit diylc.l4j.ini file and change this one flag to be -Xmx2512m instead of -Xmx512m...it will increase the limit and the file will work.
But again, I suggest against large bitmaps, whenever possible.

ryand59, 2019-02-22:
Bancika (& EL34xyz),

Thanks.  That makes sense.  I think I'll just avoid doing that -- it's not
really necessary, I was just playing around with it -- getting to know the
app.  Good to know the workaround though!

Ryan D.
----
ryand1959@gmail.com
M: 325.668.3071

EL34xyz, 2019-02-22:
Bitmap images are huge compared to png, jpg and gif
Try a png and see how that goes

I use gif's quite a bit in CAD as background images

bancika, 2019-02-22:
yeah, but the problem is that DIYLC serializes everything as raw bitmaps. Even jpeg and gif.

ryand59, 2019-02-23:
bancika,

It's academic at this point since I don't "need" those files and I don't foresee importing jpegs into drawings in the future, but I thought I'd let you know that I tried the change you suggested to the .ini file...

First, I had a file system permission problem and it wouldn't let me save the edited file.  Got that sorted out, made the change, and then had a "dead on arrival" problem when I tried to launch the app -- nothing would happen.  I changed the parameter in the .ini file back to the original parameter and it fired right up.

Please, this is not a priority or even needed!  Now that I know what caused the issue, I will avoid the problem.  Just trying to keep you informed.

The app is wonderful, btw.  I was able to document my guitar's wiring before I ripped it apart and modded the heck out of it.  I did so in very little time with absolutely no reading any documentation or watching any tutorials.

Nice work!

Is there someplace I could learn about how to make a component?   As part of my guitar's mods, I put a new pickup selector switch in my guitar (a Freeway Ultra) and if I'm able, I'd love to contribute to the tools for others to use.  Let me know if it's possible and what it would take to do.  I'm not really a coder, so if it requires writing any java or python or anything like that, I probably couldn't do it. 

Thanks for the tool and for the support!

Ryan D.

bancika, 2019-02-24:
Sure, but it's good you reminded me about it, it's really embarrassing how I handle images :)

I made a change for the next version that will make much smaller files and require less memory to work with images. Keep an eye for 3.54 :)

EL34xyz, 2019-02-24:
That's good news because I like to add background images for amp layouts in order to create a board with the terminal holes in the same location.
It's nice to add a .gif board image as a background for situations like that
I do it in a 2D CAD program now, but doing it in DIYLC would be great

EL34xyz, 2019-02-24:
Also, I had posted here a long time ago that there was an issue in adding an image as a background in DIYLC

If I create a 10 inch image in my image app and then add it to DIYLC, the image is about 1/3 the original size.
So the image is not at the proper scale it was created
I will try and find that post I created again because it had the exact details

aka13-404, 2023-02-21:
> Yes, that's it. DIYLC is not really optimized for handling bitmaps and may consume more memory than needed. Also, files are huge because it uses inefficient way of serializing the data. Your image [...]

How would I go about increasing memory available in more recent versions? I can't find any ini files. 
I have more ram than I need, and I have a lot of 512x512pngs in a layout. I can't export a png, and run out of memory.