bancika/diy-layout-creator #438: version 54 image issues

EL34xyz, 2019-02-26:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

EL34xyz, 2019-02-26:
I am testing version 54 by adding images
Note that this same thing happened on older versions

I added a .png image that is 9.389 inches wide x 3.861 inches tall
It is a 72 pixels per inch PNG image

After adding it to DIYLC
It displays in DIYLC as 3.38 inches wide x 1.39 inches tall

How come DIYLC shrinks the image down so much?
You can't even move the size slider far enough to get the image up to the original size

Here's a couple screen shots
One is from my image editor program - Paint Shop Pro
The other one is how it displays in DIYLC

[image]

[image]

EL34xyz, 2019-02-26:
Here's another issue I ran into when adding images

After clicking on the image icon to load an image, I get an error dialog box
The image does load and then the dialog box goes away

Then I added some text below the image and edited the text
That produced the error dialog box also but the edited text did display and the box closed

Here's a couple screen shots
One is while loading the image
The other one is after I edited the text below the image

[image]

[image]

bancika, 2019-02-26:
can you upload the logs please?

EL34xyz, 2019-02-26:
Ok, the error message boxes are gone now when I add an image and add/edit text under the image

The image size is still goofy
You have to multiply the imported image size x 2.77 to get the real image size of 9.39 inches
The image slider only goes to 254% which is not enough to get the image to the proper scale

bancika, 2019-02-26:
It's the difference in DPI. Your file is 72dpi while DIYLC draws in 200dpi. That's where the 2.77 comes from.

Not sure how to deal with it because file DPI can be anything, not just 72dpi. I could pre-scale by 2.77 but some other files might get wacky.

EL34xyz, 2019-02-26:
Ok, did not know it was 200 dpi

Most images people are going to use are the standard 72 dpi
If you take a screen shot from a PDF, those are usunally 96 dpi

Also, a 200 dpi image file is way larger in size than a 72 dpi file
I don't think I have ever created or used an image more than 96dpi

Maybe a wider slider range would help get the image to the proper scale?
300% would do the trick for 72 dpi images

EL34xyz, 2019-02-26:
Jut as a test, I took that same Fender layout diagram that was 72 dpi and converted it to 200 dpi
It came out the correct size in DIYLC

bancika, 2019-02-26:
yup

EL34xyz, 2019-02-26:
How about the 300% slider to adjust the size?

Many people do not know how to create a 200 dpi image
Most of the amp guys will be getting layout diagrams from a web page and those images will be 72 dpi images

daturach, 2019-02-27:
Faced that issue as well, did not know exactly where it was coming from. I used Inkscape to resize at the original dimension and imported into DIYLC.

EL34xyz, 2019-02-27:
I find it hard enough to get people to do even the simplest of computer operations.
So telling someone to convert images to 200 dpi is next to impossible for most people I deal with.

bancika, 2019-02-28:
ok, will increase the range to 500%

closed 2019-02-28 by bancika