bancika/diy-layout-creator #439: Uncaught exception in main method: java.lang.NullPointerException

lazerhawley, 2019-02-26:
I've hunted around for an hour or so for a solution to this but every time I open DIYLC I get a java.lang. NullPointerException error. I tried rolling back to Java 6 and reinstalling the current Java altogether, I re-downloaded DIYLC, and tried everything I ran into online but I haven't had any luck. I'm wondering (hoping) there's a quick and easy fix that I'm just too stupid to know about. Thanks!

bancika, 2019-02-26:
did you install the latest 3.54 version? This is a known issue in 3.53, but it should be resolved.

lazerhawley, 2019-02-26:
It was 3.54. Thanks for getting back to me so quickly! I’m really looking forward to the program.

bancika, 2019-02-26:
Oh man, I fixed one issue and caused another. Please install 3.52, then
upgrade to 3.54 and it will work. I'll fix the issue for the next version.

lazerhawley, 2019-02-26:
Fantastic! Thank you.

bancika, 2019-02-26:
I just updated 3.54 build with the fix for that too...so you can just
upgrade :)

closed 2019-02-26 by bancika