bancika/diy-layout-creator #440: Hangs on edit project settings

MurrayVK3, 2019-02-28:
The latest version 3.54.0 hangs when selecting "Edit Project Settings" from the edit menu or right click menu.

I downloaded diylc-3.54.0.zip, deleted ~.diylc (for a fresh start), executed 'run.sh' and selected "Edit project Settings".  The program become unresponsive.  Repeatable - does this every time.

OS is 64 bit Linux (Mint 18), Java is 1.8.0_171
Log file attached.

Thanks!

[log-2019-03-01 07.40.11.txt](https://github.com/bancika/diy-layout-creator/files/2916814/log-2019-03-01.07.40.11.txt)

MurrayVK3, 2019-02-28:
I went back to the previous version I was using: 3.45 and Project Settings was fine.  So I started downloading and testing releases;  version 3.46 is OK, 3.47 fails to open Project Settings  so the regression happened in there somewhere.

Hope that helps.

MurrayVK3, 2019-03-07:
New information:  The hang is only temporary, about a minute.  A Java thread is 100% busy for this time then the Edit Project window appears.    Thereafter I can open Project Settings without the hang until I close DIYLC and run it again.

I also get the same effect on editing some selections, text label in particular.  The first time after starting DIYLC I have to wait a minute for the edit window to appear.  Then its fine.

Could this be my java installation perhaps?  I don't seem to have the same problems with any other Java programs but then I don't use that many.

Thanks.

MurrayVK3, 2019-03-07:
Ah, so this is a dupe of #372

bancika, 2019-03-07:
This has to do with font initialization...I couldn't find a way round it. It will initialize on its own if you leave the app running for a minute or two before attempting to edit anything that contains a font.

closed 2019-04-01 by bancika