bancika/diy-layout-creator #441: Tracks covered by shapes highlighted

SlyWink, 2019-02-28:
Hello,
First of all, great piece of software.
I consider using shapes and dots to create specific components for my projects.
When I probe a Vero Board using the Highlight Connected Areas functionality, all the tracks covered by a shape are highlighted.
According to me, a shape is not a conductive material and thus should not interfere with tracks.
Am I wrong or is this a bug ?
SL

bancika, 2019-02-28:
a quick test shows that it works as expected, I have two traces and a shape that joins them, but highlight catches only one side at a time. Do you have an example?

[image]

SlyWink, 2019-02-28:
Here is an example :
[image]

bancika, 2019-02-28:
hmm, are you using one of recent versions? I can't reproduce and don't see anywhere in the code what might cause it, Here's the same example

[image]

SlyWink, 2019-02-28:
This is strange.
I have created a new board like yours and the behavior is as expected.
But the trouble still exists in my previous test (si attached zip file)

[testdiy.zip](https://github.com/bancika/diy-layout-creator/files/2917336/testdiy.zip)

bancika, 2019-03-01:
I got it. It has nothing to do with shapes. There's a bug with copper traces that makes any component that gets rendered after them conductive :) If you remove copper trace from the layout it will work fine. I'll fix for 3.55

closed 2019-03-01 by bancika