bancika/diy-layout-creator #446: Feature Request: Separate Visible Grid and Snap to Grid sizes

JollyJimmy, 2019-03-05:
Having a small snap to grid size is essential but the small visible grid that accompanies it is not useful as a solid reference for seeing things at a glance. 

Being able to have a grid of a particular size for a quick and easy visible reference and a separate smaller snap to grid size for precise placement relative to the larger visible grid would be a great addition.

This could be implemented in a number of ways, from a bold large grid along with a faded smaller "snap to grid" grid, or a large grid as a simple dot every division and then squares for the snap to grid subdivisions. Or perhaps even visible dots or squares and an invisible "snap to grid" grid.

JollyJimmy, 2019-03-12:
[image]

Here's a example with a dotted grid reproducing the 100 thou divisions in the top and side ruler bars and the square snap to grid set to 25 thou. I currently use this as a template using dots from the connectivity menu, but it would be a brilliant to have a native option. The setup for this workaround is a little tedious but overall it saves me a lot of time. Makes part placement that little bit easier and faster.

bancika, 2020-03-27:
How about this solution. I added "Dot Spacing" property that is expressed in grid spacing units. When set to 1 (default), nothing happens. But you can set it to a positive integer number and then it becomes active.

[image]

bancika, 2020-03-30:
this feature will go out with 4.1.0 version. Hope you'll like it :)

closed 2020-03-30 by bancika

FentonBresler, 2020-03-30:
This looks like just the thing, wow! Many thanks!