bancika/diy-layout-creator #448: Switch body size not changing

EL34xyz, 2019-03-08:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

EL34xyz, 2019-03-08:
I pasted a mini DPDT toggle switch
Then I changed the pin spacing from .2 inches to .25 inches
The pins expanded correctly to a .25 inch grid spacing,  but the switch body size did not expand

Check out my screen shot

[image]

EL34xyz, 2019-03-09:
I figured out how to make the switch body expand or contract after changing the pin sizes.
After you change the pin spacing, grab the switch and move it
Thee body will snap to the correct size relative to the pin spacings

bancika, 2019-03-09:
Yup, it's only the spacing that doesn't update the size, I'll fix that

EL34xyz, 2019-03-09:
Ok, thanks
I went back to play around with switches again today and found that moving it around thing by accident :)

closed 2019-03-10 by bancika