bancika/diy-layout-creator #451: Can't add components as a separate jar file

JiriBilek, 2019-03-11:
Hi,
thanks for the app, it is quite useful. As I plan to use it for documenting my microcontroller projects, I wanted to add some components (connectors, pin headers, for instance).

- I am using Java 1.8.0_181 on Windows 7 and recent version of Eclipse.
- I cloned diy_layout projects successfully and compiled all three of them (_diylc-core_, _diylc-library_, _diylc-swing_).
- I am able to run the project from Eclipse (and see the application log).
- I created a new component as a part of _diylc-library_ project and is working fine.
- I created my own project and moved the new component classes there, compiled and created jar file with only my 2 classes in it. I copied the jar file to _diy-layout-creator/diylc/diylc-swing/library_ folder (there had been one file so far - _main,jar_).

But now, I am not able to see my new component in the application, it seems to me that my jar is not read by the classloader. I renamed the package name of my component to _org.diylc.components_ but it didn't help.

Partial dump of the log:

> ...
> 21:01:00,773  INFO [main] Presenter - Checking if variants need to be updated
> 21:01:00,784  INFO [main] Presenter - installPlugin(ToolBox)
> 21:01:01,030  INFO [main] Presenter - Loading component types.
> 21:01:01,031 DEBUG [main] Utils - Scanning jar: /D:/Prace/Eclipse_Workspaces/diylc/diy-layout-creator/diylc/diylc-swing/library/main.jar
> 21:01:01,216 DEBUG [main] Utils - Scanning directory: D:\Prace\Eclipse_Workspaces\diylc\diy-layout-creator\diylc\diylc-core\bin\org\diylc\components
> 21:01:01,342 DEBUG [main] Presenter - Guitar: [Single Coil Pickup, P- Bass Pickup, LP Toggle Switch, Lever Switch, Humbucker Pickup, Jazz Bass Pickup, P-90 Single Coil Pickup]
> ...

It seems to me that my jar file (_jb-library.jar_) is not being searched. Because the code that should read the jar is in another project (function org.diylc.appframework.miscutils.Utils.getClasses), I was not able to trace it.

I would be very glad if you can help me as the magic of dynamically loading classes is rather unknown to me. Maybe it's my bad, I tried to follow instructions in file _diy-layout-creator/ComponentAPI.md_ as far as I could.

My jar (zipped):
[jb-library.zip](https://github.com/bancika/diy-layout-creator/files/2954096/jb-library.zip)

bancika, 2019-03-14:
Seems like I made it scan only the jars that are already on the classpath, so it ignored your jar. I fixed that and pushed the change. Now your jar gets scanned but fails because it seems to be built with Java 8 and the rest of DIYLC is Java 6.

Out of curiosity, why don't you add the components to the main package? Could they be potentially useful to wider audience?

JiriBilek, 2019-03-14:
Thanks, it works now even in Java 8 (after recompiling all parts of the app).
I plan to design some parts for general MCU prototyping (pin headers, buttons, connectors). Once it is ready I will create a PR.

The reason I am working on a separate library for now is compiling speed. It seems to me that ant scripts take a long time to finish. As if they always recompiled the whole project. I am not an expert in ant (to be honest, this is my first encounter with ant) and I am not able to tell the ant to just recompile changed files.

Closing because the issue was fixed, thanks.

closed 2019-03-14 by JiriBilek