bancika/diy-layout-creator #455: components and solder pads dont stay on top. on 3.57, and when you copy them they dont keep theyr name

baddriddim, 2019-04-14:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

bancika, 2020-03-27:
do you have an example for the first issue?

bancika, 2020-12-11:
closing due to inactivity. Please comment with more info to reactivate.

closed 2020-12-11 by bancika