bancika/diy-layout-creator #456: when you copy a group, it's not pasted properly, it appears really weird, make things complicated

baddriddim, 2019-04-14:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

bancika, 2020-03-31:
what do you mean? Do you have an example? A diy file or a screenshot would help.

bancika, 2020-12-11:
closing due to inactivity. Please comment with more info to reactivate.

closed 2020-12-11 by bancika

htor, 2020-12-12:
@bancika 

i think the github issue template is confusing for some. i suggest wrapping the text ("Thank you for reporting issues...") in a html comment `` so it does not get copied into the actual issue without user interaction. above the comment it could also say `OS: <your os>, DIYLC version: <your version>` so you don't need to ask that. cheers!