bancika/diy-layout-creator #457: DIP numbering mirroring

julessouren, 2019-04-17:
If a trace-side view of a board with a DIP IC is created (i.e. a mirrored view of the component side), then the pin numbering on the IC is not mirrored.

In other words, a DIP IC cannot be modified such that it appears to be viewed from through the board, and have its pin numbering shown correctly (mirrored).

I find this feature useful for debugging boards, by immediately identifying correct pin numbers on the trace side, without having to think mirrored.

bancika, 2020-03-25:
would it be sufficient to add two more options here called "DIP (Mirrored)" and "Connector (Mirrored)" that would show the numbering mirrorer?

[image]

bancika, 2020-04-19:
added for 4.5.0

closed 2020-04-19 by bancika