bancika/diy-layout-creator #458: Highlight Connected Areas tool is not working properly.

starpath07, 2019-04-19:
[log-2019-04-19 04.16.36.txt](https://github.com/bancika/diy-layout-creator/files/3098504/log-2019-04-19.04.16.36.txt)

Highlight tool is showing unrelated (and unconnected) stripboard tracks as connected. Note the attached screen grab where I have removed the link between rows V and W. The ground plane should stop being highlighted from row W downwards. This is a proven design which I have built and is functioning properly. 
[image]

jovi2k, 2019-05-04:
It seems that when a jumper length is even, the connected points are not only the jumper ends, the middle point is also connected with them. This is not an issue with hookup wire, so you can edit hookup wire with two points and save it as variant and use it afterwards.

Jumper with even length
[image]

Jumper with odd length
[image]

Hookup wire with even length
[image]

Hookup wire with odd length
[image]

csj2k, 2019-07-30:
I also have this issue.

bancika, 2020-03-27:
found what it is...will be included in the next version.

closed 2020-03-27 by bancika