bancika/diy-layout-creator #459: Is there a way to set dimensions of boards or shapes?

dorrisant, 2019-04-19:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

dorrisant, 2019-04-20:
The main reason for this request is for aligning board mounting holes with existing chassis holes. If the chassis holes form a rectangular pattern, it can be reproduced by making a rectangle and setting the dimensions. This rectangle can be superimposed over a board and the corners will show exact placement of board holes. It (the rectangle) can then be moved around to find the best placement for board mount hardware. The holes can be placed on the board, then the rectangle can be deleted to clean things up.

For the same reasons this method can be used for precise spacing of all hardware like tube sockets, pots, jacks, switches, etc.. I export board layouts into Corel to facilitate the transfer of DIYLC files to be produced on the CNC.

EL34xyz, 2019-06-07:
Several of the CAD and CAM programs I use all allow you to enter exact lengths of objects in a dialog box
I have been wanting something like this in DIYLC also
It could be added to the Edit selection dialog box

So for example, you draw a line and you want it some length other than what the grid is set to
Here's some of the thing that could be set in the edit selection dialog box

Set the X,Y start points
Set the X,Y end points
Set the exact length of the object
On rectangles set the width and height
On ellipse or circles set the diameter or radius

[image]

bancika, 2019-06-07:
It can be done but you can't have both control points there *and*
width/height properties exposed. Only one control point can exist if you
are specifying explicit dimensions, as shown on your screenshot.
In DIYLC, the only way would be to allow for shapes to have two modes of
operation, one as it is now and one with one control point and editable
size.

EL34xyz, 2019-06-07:
Just showing examples of what other software lets me set

The handiest things I use all the time are in other software is
Setting a lines exact length
Setting a rectangles height and width
Setting a circles diameter

EL34xyz, 2019-07-13:
This would be really handy to set a boards exact dimensions rather than dragging and wishing you could get it exact.

bancika, 2020-03-26:
Added for 4.0.1

closed 2020-03-26 by bancika