bancika/diy-layout-creator #463: osx.zip file damaged

sthcoaster, 2019-05-29:
Just discovered your app today on a thread at DIYAUDIO. I'm just planning an after school programme for middle school - this app will be a great resource. Thanks for your efforts.

In OSX Mojave I "Allow App Store and identified developers" in preferences.
Latest JRE installed.

Osx.zip download unpacks and icon for DIYLC presents with no difficulty
When click icon receive the following msg

"“DIYLC.app” is damaged and can’t be opened. You should move it to the Trash."

Thanks

dbeckham, 2019-05-31:
This is also happening to me. (macOS Mojave 10.14.4)

Lycoming360, 2019-06-07:
Also to me:   Mac OS Mojave 10.14.5

bancika, 2019-06-07:
It's definitely not damaged. One of apple's nonsense moves. There are few
things that can be done but I'm not a mac user. There is tons of info
online.

sthcoaster, 2019-06-08:
OK 
This worked for me
https://apple.stackexchange.com/questions/243687/allow-applications-downloaded-from-anywhere-in-macos-sierra

closed 2019-06-08 by sthcoaster

bancika, 2019-06-08:
Great!

bancika, 2020-03-22:
Can anyone with Mac OS give this version a try? I tried it on Mojave VM and it works fine, but it dies after starting on Sierra VM. Can't figure out the reason yet.

[diylc-4.0.1-osx.zip](https://github.com/bancika/diy-layout-creator/files/4365497/diylc-4.0.1-osx.zip)

sthcoaster, 2020-03-23:
Yep works fine for me on Mojave - needed to download latest Java 8

bancika, 2020-03-23:
Awesome, thanks. Here's another one with a new icon :)

[diylc-4.0.1-osx.zip](https://github.com/bancika/diy-layout-creator/files/4369400/diylc-4.0.1-osx.zip)

sthcoaster, 2020-03-23:
No bother. 

Used your program to layout this Starving Student Miller Hybrid (12AU7 version) together. Like many I will be using the time covid has given me to finish this (and other projects) off over the next weeks!!

Thanks for your support of diy

Clayton

cenobitedk, 2020-04-19:
Do this
https://www.youtube.com/watch?v=6fqzb4qpgcs
or this
https://www.youtube.com/watch?v=3py0kgq95Hk