bancika/diy-layout-creator #464: Features request: auto save and wire length

jonrawpower, 2019-06-06:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

bancika, 2019-06-06:
hey, auto-save is already there.  DIYLC keeps a history of file edits in your user folder, inside .diylc subfolder.

EL34xyz, 2019-06-07:
I think what you are asking for is the same as in this post
https://github.com/bancika/diy-layout-creator/issues/459

bancika, 2019-06-07:
doesn't seem to be the case here.

EL34xyz, 2019-06-07:
I was thinking a wire length setting is like setting a lines length?

bancika, 2019-06-07:
I understand he wants to get the total wire length of a curved wire once
it's set using control points, not to enter the length. We can't apply
sizing to curved lines as you really need all the points to set the
position exactly.

bancika, 2020-03-26:
Added wire length feature for version 4.0.1

[image]

closed 2020-03-26 by bancika