bancika/diy-layout-creator #468: grid problems

baddriddim, 2019-07-26:
I need to be able to trace a line in the middle for a rackbox faceplate design, i'm using milimeters, i need to point a 242 mm, and i set my grid at 1mm spacing but there are 12 lines between 240 and 250 mm, and the vertical line does not even get on the line properly, at each line the measures diplayed are weird numbers, never exact. 
Thnak you for your work and attention, i'll send another 10 next month. i love this software! but to be honest since the latest version there is a lot of loss of time, every time you need to put things in front or in back because it doesnt do it automatically, what goes on top is always the last put in place, but a solder pad should Always be over a line imo. maybe its convenient for multi layer or whatever but when working on one layer its a damn loss of time with manipulations.
just to let you know, i am now getting big printed platic stickers for my rack faceplates (the kind of big size stickers you can see used on vehicules for adds), i send the faceplate layout to my clients in black and white, and then they can design them however they want, matching the colors to theyre sound systems logo or rig colors. they love it. 
thank you for this great simple and worthy software, i love it, i'll send you sum money more regularly when i make sells and ask my clients who use it to send one or two buck for the use
greetings from france
 Sebastien (baddriddim workshop)

bancika, 2020-04-13:
do you have a screenshot as an example, it's a bit hard to follow.
Thanks

bancika, 2020-12-02:
this should be fixed

closed 2020-12-02 by bancika