bancika/diy-layout-creator #472: enhancement suggestion -- coloring 1 line in veroboard (GND or POWER rail)

peterjosvai, 2019-08-17:
hello, Branislav, 
I admire your work, thank you! 

one suggestion:
it would be great if one line of a veroboard could be given a color, 
like black for a "line" (a strip) dedicated to GROUND, for example...
or RED for a DC POWER rail +, you know what I mean :)

apart from having seen hundreds of veroboards created by DIYLC, 
I've been actually using it since last night only,
so, if this feature already exists, pardon me, please :)

pixelwaster, 2019-12-24:
Like this - 
[image]

You might have to turn off **Config -> Snap to Grid**  depending on your grid size, Just makes it easier to place the box over the trace.
Use the **Shape -> Rectangle** on the tree (or tabbed toolbar) and put two boxes down
Select one of them and size to fit over your power rail.
Bring up the **Edit Selection**  dialog (double click, ctrl+E, **Edit->Edit Selection**)
[image]

Salt to taste, but 36% and FF0000 is a good start.
Repeat for the ground rail.
Select the three parts and group them. (ctrl+G or **Edit->Transform Selection->Group Selection**)
Turn **Snap to Grid** back on if you turned it off.
You could also save this as a building block.

bancika, 2020-03-25:
thanks @pixelwaster

closed 2020-03-25 by bancika