bancika/diy-layout-creator #473: desktop launcher (in Linux)  doesn't work: Error: Could not find... org.diylc.DIYLCStarter

peterjosvai, 2019-08-29:
hello, 

I use Linux Mint, and I run DIYLC from a directory, where it is placed as "diylc-3.28.0"...
in which I will run "run.sh"...

when I click on run.sh, it works, giving me the dialogue if I want to run it in terminal or want to view it ("display"), or "cancel", or just "run"...

BUT, as soon as I create a desktop launcher...
it won't start...
note: owners rights are all right :)

so, the point: 
when run the launcher it runs this:
"/path-to-directory/run.sh"

if I run this ("/path-to-directory/run.sh") in that directory ("/path-to-directory"), it works...

if I run this in the directory "/desktop", it doesn't...

and in this case it will display this (in the terminal where I was executing the command)

quotation:
Error: Could not find or load main class org.diylc.DIYLCStarter
Caused by: java.lang.ClassNotFoundException: org.diylc.DIYLCStarter
/quotation-end

ChrisOfBristol, 2019-10-22:
A bit late perhaps, but...I can let you have a .deb file which you just click on to install. It runs on Ubuntu (standard version) and Ubuntu Budgie 19.04 and will probably run on Mint too.

hexeguitar, 2019-12-15:
I had the same problem while creating a system wide alias to start DIYLC from the command line.
The solution was to update the java run command present in the run.sh script with the correct path to the directory where the diylc is installed.
This is the content of the run.sh script:
`java -splash:splash.png -Xms512m -Xmx2048m -Dorg.diylc.scriptRun=true -Dfile.encoding=UTF-8 -cp diylc.jar:lib org.diylc.DIYLCStarter`

Let's say the path to the DIYLC files is `~/Software/DIYLC/`

The new command will look like this:
`java -splash:~/Software/DIYLC/splash.png -Xms512m -Xmx2048m -Dorg.diylc.scriptRun=true -Dfile.encoding=UTF-8 -cp ~/Software/DIYLC/diylc.jar:lib org.diylc.DIYLCStarter`

Of course, the path to the splash.png and the diylc.jar flag should be updated to the one you have.
Instead of pointing to  and executing the the run.sh script you can use the above command as a launcher in the desktop file.
In my case i have made the launcher as a `diylc` file put into `~/bin/` directory with a small addition to make it run in background and detach from the current terminal window:
`nano diylc`
```
#! /bin/bash -e
java -splash:~/Software/DIYLC/splash.png -Xms512m -Xmx2048m -Dorg.diylc.scriptRun=true -Dfile.encoding=UTF-8 -cp ~/Software/DIYLC/diylc.jar:lib org.diylc.DIYLCStarter </dev/null &>/dev/null &
```
Save & exit, make the file executable:
`chmod +x diylc`
To update the PATH with the new command:
`source ~/.profile`
This way i have a system wide command to run the DIYLC from any place being in a terminal. Having that simply putting "diylc" as a command in the desktop launcher will work, too.

Hope that helps!

peterjosvai, 2019-12-15:
thanks guys, both of you! 
hexeguitar, this is what I needed -- what everybody needs, I guess :) 

thanks a lot!
 ChrisOfBristol , you too!, thanks!

closed 2020-03-25 by bancika