bancika/diy-layout-creator #474: diy-layout-creator is only backward compatible till DIYLC Gen3 v55.1 only. (not v56 or v57)

MadiI-Hup, 2019-09-25:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

bancika, 2019-10-03:
Hi,
do you have an example? Otherwise it could mean anything.
Thanks

MadiI-Hup, 2019-10-06:
Hi
Please find attach is a sample layout that opens normally with v. 55.1 and earlier.
If you want an illustrated pictures, let me know.
By the way, thank you for the great product.

Thanks
madi

bancika, 2019-10-06:
hmm, couldn't find the file, can you please try to upload it through github (vs though email)?

MadiI-Hup, 2019-10-06:
Hi,
See The attached zip file

[SIO-project_mini_3a.zip](https://github.com/bancika/diy-layout-creator/files/3694520/SIO-project_mini_3a.zip)

closed 2019-10-06 by MadiI-Hup