bancika/diy-layout-creator #478: DIYLC v 3.57.0 will not save files on my Windows 10 laptop.

Breeze68, 2019-10-10:
This has been happening on this laptop for a while now. Though an upgrade to newest version would help. It did not. Can you help?

[log-2019-10-10 08.48.02.txt](https://github.com/bancika/diy-layout-creator/files/3712617/log-2019-10-10.08.48.02.txt)

bancika, 2019-10-10:
I see "access denied" error in the log, meaning that windows denied
permission to that folder. You can try two things:

1) run DIYLC in administrator mode
2) try to write the files somewhere else (other partition).

does any of that solve the problem?

cheers

Breeze68, 2019-10-17:
Thanks so much for this Bane.

It did not want to save files within the Documents Folder of the Windows 10 Setup.
I created a DIYLC Files Folder under the C drive heading and it worked just fine.

Much appreciated.

Love the program by the way.

Mike!

Mike Brisbin
Project Manager
MVI- MultiVision Inc.
Cell: (647) 393-9331
Office: (416) 449-1700 x234
Web: www.mvidisplay.com<http://www.mvidisplay.com/>
Email: brisbin@mvidisplay.com<mailto:brisbin@mvidisplay.com>

closed 2020-03-24 by bancika