bancika/diy-layout-creator #479: Mac OSX upgrade to Catalina upgrades Java to a version that is incompatible with DIYLC v. 53

Swampertech, 2019-10-13:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

Swampertech, 2019-10-14:
I have managed to open diylc by double clicking on the diylc.jar file which is in the zip folder.

r-i-c-h, 2019-11-16:
To be specific I had to do the following: 
1. ctrl-click/[right-click] on DIYLC.app
2. Show Package Contents
3. Contents>Resources>Java
4. Open the aforementioned diylc.jar

r-i-c-h, 2019-11-19:
I want to add a link here for a thread that lets you download a compatible version of Java to Mac OS - https://forums.macrumors.com/threads/java-6-runtime-legacy-version-for-macos-catalina.2195954/
The 2nd page even has some DIYLC specific actions.

foggy09, 2019-12-11:
Hi I am setting up anew MacBook Pro with Catalina and want to bring across my DIY-LC but have read there is problems with Java and DIY installation. Having read the above this is the message I get when trying to install Java as the Mac says it has a newer version. Need help

peterdfolk@Peters-MBP ~ % set theDMG to choose file with prompt "Please select javaforosx.dmg:" of type {"dmg"}
do shell script "hdiutil mount " & quoted form of POSIX path of theDMG
do shell script "pkgutil --expand /Volumes/Java\\ for\\ macOS\\ 2017-001/JavaForOSX.pkg ~/tmp"
do shell script "hdiutil unmount /Volumes/Java\\ for\\ macOS\\ 2017-001/"
do shell script "sed -i '' 's/return false/return true/g' ~/tmp/Distribution"
do shell script "pkgutil --flatten ~/tmp ~/Desktop/Java.pkg"
do shell script "rm -rf ~/tmp"
display dialog "Modified Java.pkg saved on desktop" buttons {"Ok"}
zsh: parse error near `do'

Swampertech, 2019-12-11:
Foggy,
	Here is the solution that worked for me.

http://el34world.com/Forum/index.php?topic=25176.0 <http://el34world.com/Forum/index.php?topic=25176.0>

Hope this helps.

foggy09, 2019-12-11:
Thanks Swampertech
I did what was said about Java and used Pacifist and it seemed to Install the correct Java then I tried to open DIY-LC and it came up with
[image]
I went too the security but in a Mac with Catalina it will not show me the "Allow 3rd party etc"

Swampertech, 2019-12-11:
Foggy,
	This is the fix I sent to Doug Hoffman on his site when I had this problem:

Doug, I wanted you all to know that there is a fix for the Java SE6 problem incurred after upgrading to Catalina on Macs. There is a programmer, bogdanw  , over on the MacRumors forum that provided a script you can run to allow a legacy version of Java to run. I was able to download and install that Java version which is  compatible with DIYLC. After upgrading DIYLC to v. 3.57, I couldn't open my old v. 3.53 files until I had given DIYLC full disk access. I hope this will help any Mac users out there running DIYLC.

If you will directly contact bogdanw over at MacRumors site, he can walk you through the fix. Hope this helps.

Swampertech

r-i-c-h, 2019-12-19:
Here are my DIYLC Mac/Java Fix Instructions cobbled together from this [macrumors thread](https://forums.macrumors.com/threads/java-6-runtime-legacy-version-for-macos-catalina.2195954/) and this [el34world thread](http://el34world.com/Forum/index.php?topic=25176.0). I borked things up so many times trying to get stuff to work that I "removed" Java, but I don't know that that is a necessary, however I've included those instructions at the bottom just in case...

# Installing Java and DIYLC

1. Download "Java for OS X 2017-001" from https://support.apple.com/kb/dl1572?locale=en_US
2. Note where you save that .dmg file (probably in your Downloads directory) because we'll need to find it later. 
3. Open up the Script Editor.app (it's in Finder's "Go" menu-> Utilities)
4. Copy and paste the following into the script editor window: 

```
    set theDMG to choose file with prompt "Please select javaforosx.dmg:" of type {"dmg"}
    do shell script "hdiutil mount " & quoted form of POSIX path of theDMG & " -noverify -nobrowse"
    do shell script "pkgutil --expand /Volumes/Java\\ for\\ macOS\\ 2017-001/JavaForOSX.pkg ~/tmp"
    do shell script "hdiutil unmount /Volumes/Java\\ for\\ macOS\\ 2017-001/"
    do shell script "sed -i '' 's/return false/return true/g' ~/tmp/Distribution"
    do shell script "pkgutil --flatten ~/tmp ~/Desktop/Java.pkg"
    do shell script "rm -rf ~/tmp"
    display dialog "Modified Java.pkg saved on desktop" buttons {"Ok"}
```

5. Run the script by clicking on the play arrow at the top of the window.
6. A window will pop up asking you to select the `JavaForOSX.dmg` file from step #2
7. Choose it and wait for the magic to happen. The Script will put a file called `Java.pkg` on your desktop. 
8. Close the Script Editor, and then double-click the new Java.pkg on your desktop to install as you normally would...
9. Download DIYLC from [Github](https://github.com/bancika/diy-layout-creator/releases)
10. UnZip and then drag the DIYLC.app icon into your Applications folder.
11. With the app in /Applications, open Terminal and then copy/paste and enter the following: 
    * `sudo xattr -rd com.apple.quarantine /Applications/DIYLC.app` 

*Note* - After a `sudo` terminal command, the terminal will ask for you to type in your system password. It will look like the terminal isn't allowing you to type anything in as there won't be any *s or anything showing up, but it *is* working. Just hit enter when you are done even though it doesn't look like the terminal responded to your typing.

12. Open the Applications folder if you've closed it, then "right-click" (hold down the option key and click) on DIYLC.app and select "Open".
13. (Trouble With file access? Open System Preferences --> Security & Privacy --> Privacy --> Full Disk Access and add DIYLC.)

Hopefully you're good to go...

---- 

# Removing Java
Don't think this is necessary, but it was part of my process before I did the above, so figure I should include the info...
1. Following instructions from the Java website's ("How do I uninstall Java on my Mac?")[https://www.java.com/en/download/help/mac_uninstall_java.xml], sequentially copy and paste each of these lines into the terminal and hit enter. 
      1. `sudo rm -fr /Library/Internet\ Plug-Ins/JavaAppletPlugin.plugin`
      2. `sudo rm -fr /Library/PreferencePanes/JavaControlPanel.prefPane`
      3. `sudo rm -fr ~/Library/Application\ Support/Oracle/Java`

2. [This article](https://explainjava.com/uninstall-java-macos/) adds `sudo rm -rf /Library/Java/JavaVirtualMachines` to the above

3. There are instructions I found on the Java website that didn't apply to my machine/weren't there but I'll mention...
    1. [How do I remove the Java Deployment cache after Java is uninstalled?](https://www.java.com/en/download/help/deployment_cache.xml)
        * I copy/pasted `rm -r ~/"Library/Application Support/Oracle/Java"` but I got a "no such file" error
    2. [How do I clear the Java cache on my Mac?](https://www.java.com/en/download/help/mac_java_cache.xml) 
		* the icon wasn't in my System Preferences
    3. From [Uninstalling the JRE](https://docs.oracle.com/javase/8/docs/technotes/guides/install/mac_jre.html#A1096890):
        * Navigate to /Library/Internet Plug-Ins and remove the JavaAppletPlugin.plugin directory. [Wasn't there]
        * Navigate to /Library/PreferencePanes and remove JavaControlPanel.prefpane. [Wan't there]
   4. From (Uninstalling the JDK)[https://docs.oracle.com/javase/8/docs/technotes/guides/install/mac_jdk.html#A1096903]:
        * Navigate to /Library/Java/JavaVirtualMachines and remove the directory whose name matches the following format: `/Library/Java/JavaVirtualMachines/jdkmajor.minor.macro[_update].jdk` - [None of the directories matched that pattern]
    5. I followed instructions to download and execute the [Java Uninstall Tool](https://www.java.com/en/download/uninstalltool.jsp) but it didn't find any Java installations on my computer

Hope it helps!

r-i-c-h, 2019-12-19:
@Swampertech - Now I can get DIYLC to open without a problem, but once I place a component in the workspace, I can't select it! (the mouse arrow never turns into a hand). Is this an issue for you?

Swampertech, 2019-12-19:
Rich,
	I don’t have that problem. Sorry!

foggy09, 2019-12-19:
Hi Rich
Thanks for all your help.
I have a problem though when I run the script and choose the JavaForOSX.dmg the script editor come up with "Error encountered while creating /Users/peterdfolk/tmp. Error 17: File exists. I have cleaned out and uninstalled anything to do with Java as explained above so it should be a fresh start. Any ideas?

foggy09, 2019-12-19:
I have also downloaded the Java Uninstall and this showed that there is no Java Installed

r-i-c-h, 2019-12-20:
@foggy09 - I can't help much, but according to the first thread, what you should do is delete the /tmp folder and basically retry from the beginning. (The instructions are there on how-to). The /tmp folder has a failed attempt from before so you have to nuke that and then try to follow the instructions.

foggy09, 2019-12-21:
Thanks again Rich I will get this all going somehow. On the message above where are and how do I find the /tmp folder. Is it in Library or do and how do use Terminal.

foggy09, 2019-12-29:
Hi All I have tried all over the Christmas to get DIYLC running on Mac Catalina but come up with the below and can't continue on.  I have uninstalled Java by using code sudo rm -fr /Library/Internet\ Plug-Ins/JavaAppletPlugin.plugin. sudo rm -fr /Library/ ... and it seemed to work. I even restarted the Mac before doing the above and I also downloaded Java from the above notice "Installing Java and DIYLC" as well as a new DIYLC App then followed the above to the letter but I still come up with the below. What am I doing wrong. I do not want to get RID of DIYLC as I have a few projects I am in the process of finishing. I am very Frustrated.
[image]

foggy09, 2020-01-12:
After Hours and Days trying to get DIYLC running on Catalina I have finally given up and will no longer have the option to use this good software. Very sad. 
Thanks to all that have made suggestion and help. Much appreciated.

proddy, 2020-01-12:
I use DIYLC on Win10 and have a MacBook with Catalina. I'll see if I can get it working...

proddy, 2020-01-12:
@foggy09 Ok, so as Swampertech mentioned in the first post, I downloaded the diylc-3.57.0.zip, unpacked it, double-clicked the JAR file, takes you to Oracle to install JDK 13 and was up and running on Catalina in less than a minute. That could be solution no?

foggy09, 2020-01-13:
Thanks for the feed back I will give this a try. Did you have to follow all the instruction as I did as was told but it didn't work. I just tried to open DIYLC and it said to to go to more info. I am not sure I tried this but will give it a go and advise.

proddy, 2020-01-13:
yes running the package/dpk doesn't work. Just install any Java SDK and launch the .jar file directly. The correct way would be to create a launch script with the correct memory settings as in the run.sh file provided.

r-i-c-h, 2020-01-17:
So, I can only update folks that my issue with "not being able to select anything" was because under the "Config" menu, there was a check beside "Highlight Connected Areas". Unchecking that allows me to move and edit components.

bancika, 2020-03-22:
Can anyone with Mac OS give this version a try? I tried it on Mojave VM and it works fine, but it dies after starting on Sierra VM. Can't figure out the reason yet.

[diylc-4.0.1-osx.zip](https://github.com/bancika/diy-layout-creator/files/4365495/diylc-4.0.1-osx.zip)

foggy09, 2020-03-26:
Hi Bancika
I will try it on my MacBook
Running Catalina. Any instructions to follow.
Regards

Mingara Pastoral
179 Reynolds Road
Bridge Creek Vic 3723
0457884847

bancika, 2020-03-26:
released 4.0.1

closed 2020-03-26 by bancika

foggy09, 2025-02-23:
Hi Bancika
I just downloaded and installed the new Mac Version diilc-4.40.0-osx-arm.zip and did what I had to do with security to get it to run but then it came up with “Unable to load Java Runtime Environment”. I run 2019 MacBook Pro 16”, 1tb, 16gb ram etc and Intel.
I was running I believe DIYLC 4.48.0 with no problems at all.
Did I download the wrong version as I noticed 2 I could choose from?

I am not sure what Version of Java I am running.

How can I fix this problem

Regards
Mingara Pastoral
Peter D Folk
179 Reynolds Road Victoria 
Australia 3723
***@***.***
0457884847

bancika, 2025-02-23:
> Hi Bancika <br> I just downloaded and installed the new Mac Version
> I run 2019 MacBook Pro 16”, 1tb, 16gb ram etc and Intel. <br> I was running
> I believe DIYLC 4.48.0 with no problems at all. <br> Did I download the
> wrong version as I noticed 2 I could choose from? <br> <br> I am not sure
> what Version of Java I am running. <br> <br> How can I fix this problem
> <br> <br> Regards <br> Mingara Pastoral <br> Peter D Folk <br> 179 Reynolds
> Road Victoria <br> Australia 3723 <br> ***@***.*** <br> 0457884847 <br>
> <br> <br> <br> <br> <br> <br> <br> <br> <br>

You need x86 version, not arm

foggy09, 2025-02-23:
Thanks I will give that a go.