bancika/diy-layout-creator #481: Won't run on new Linux version

ChrisOfBristol, 2019-10-22:
This used to run on Ubuntu 19.04 but won't on 19.10. These are the last few lines before it stops:
```
00:14:18,111  INFO [SwingWorker-pool-1-thread-3] UpdateChecker - 0 updates found
00:14:18,211  INFO [SwingWorker-pool-1-thread-2] UpdateChecker - Trying to download file: http://www.diy-fever.com/update.xml
00:14:19,255  INFO [SwingWorker-pool-1-thread-2] UpdateChecker - 0 updates found
```
I realise that the problem must be to do with the OS upgrade rather than with DIYLC, but what is it doing at that point, so that I know where to look?

ChrisOfBristol, 2020-03-11:
I think the program is hanging in lib/app-framework.jar/update/org/diylc/appframework/update
UpdateChecker.class or UpdateChecker$1.class
due to some problem checking the web for updates.

The program will also hang earlier in the startup process if I disconnect the Internet.

I can't look into this any deeper as the Java class files are not readable - and I don't know Java anyway.

bancika, 2020-03-25:
I just installed the latest ubuntu on a VM and DIYLC works there just fine. With or without the internet connection it starts.
Strange.

ChrisOfBristol, 2020-03-25:
I've downloaded 57a and run it with run.sh, it stops for a while at 17:45:39,537 then hangs. The terminal messages again suggest connection problems.
[image]

bancika, 2020-03-25:
After all that does it start? It's strange that it hangs the app, all those potentially long operations are done in SwingWorker thread (as shown in the log) to avoid blocking the main thread.

ChrisOfBristol, 2020-03-25:
The splash window opens, but nothing else, it just hangs there, no other windows open.

ChrisOfBristol, 2020-03-25:
The last line of the log is:
> 20:03:23,255  INFO [SwingWorker-pool-1-thread-1] CloudPresenter - Login failed

bancika, 2020-03-25:
try to comment out or remove these two entries from the config file

cloud.token
cloud.Username

ChrisOfBristol, 2020-03-25:
Removed them so that bit of config file looks like this now:
```
  <entry>
    <string></string>
    <string></string>
  </entry>
```
Last line of log:
`
20:44:53,171 DEBUG [SwingWorker-pool-1-thread-2] PhpFlatProxy - Connecting to: http://www.diy-fever.com/diylc/api/v1/getAnnouncements.php`

ChrisOfBristol, 2020-03-26:
I've tried this with no config.xml file, with just 
<map>
</map> 
in it and and with empty file. It doesn't seem to make much difference, as it always get stuck at about the same point.

bancika, 2020-03-26:
Ok, it fetches announcements regardless of configuration...cloud connection
is only used if you ever logged onto the cloud.

bancika, 2020-03-27:
I made the smallest possible app that connects to internet and fetches a page. Can you try to run it?

http://diy-fever.com/temp/testHttp.jar

Just type in few addresses and it should output something like this. If it hangs, it means it's something with your machine or java that's blocking the connection.

[image]

ChrisOfBristol, 2020-03-28:
> chris@desktop:~$ java -jar ./testHttp.jar
> URL: https://www.youtube.com/
> Opening stream...
> Opening reader...
> Read 224464 bytes
> Done!
> chris@desktop:~$

bancika, 2020-03-28:
So internet connection is ok. Can you try www.diy-fever.com? That's where update info is.
Apart from that, the only other idea is to try to install Eclipse on your machine and see where it hangs in the debugger.

ChrisOfBristol, 2020-03-28:
Not sure what you mean by update info on that page. Will investigate what Eclipse is.

bancika, 2020-03-28:
I mean, DIYLC connects to diy-fever.com to get update information.

As for eclipse, that's IDE used to develop Java apps.

ChrisOfBristol, 2020-03-28:
Haha! Obvious now. 
```
chris@desktop:~$ java -jar ./testHttp.jar
URL: https:www.diy-fever.com
Opening stream...
Error: protocol = https host = null
java.lang.IllegalArgumentException: protocol = https host = null
	at sun.net.spi.DefaultProxySelector.select(DefaultProxySelector.java:177)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1156)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:1056)
	at sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:177)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1570)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1498)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getInputStream(HttpsURLConnectionImpl.java:268)
	at java.net.URL.openStream(URL.java:1068)
	at testHttp.Main.main(Main.java:19)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:61)
Done!
chris@desktop:~$ 

```

bancika, 2020-03-28:
Try with http instead of https

ChrisOfBristol, 2020-03-28:
Have done...I edited the post but you probably don't get notified of that. Works ok.

bancika, 2020-03-29:
I see. Have no other clues other than to try with eclipse. If you have 15 minutes to spare, I can guide you through pulling the code from github and debugging. Eclipse for linux can be found here

https://www.eclipse.org/downloads/download.php?file=/technology/epp/downloads/release/2020-03/R/eclipse-java-2020-03-R-linux-gtk-x86_64.tar.gz

ChrisOfBristol, 2020-03-29:
I've already installed it. I'd be up for that a little later this evening- which I think is lunchtime for you :)

ChrisOfBristol, 2020-03-29:
Ready when you are.

bancika, 2020-03-29:
when you open eclipse, in the left side there's "Package Explorer". Right
click and go Import -> Project from Git

[image: image.png]

Then "Clone URI" paste this address to URI...don't need to populate other
fields.

https://github.com/bancika/diy-layout-creator.git

Then you can clone the repository and import projects into the workspace.

When done, it should look like this

[image: image.png]

Find DIYLCStarter and right click to start debug

[image: image.png]

you should see log in the bottom Console section. When it  hangs, click the
suspend button to pause the execution

[image: image.png]

and it should jump to this section that tells you where it stopped. Please
take a screenshot of this section.

[image: image.png]

Hopefully that should be enough to pinpoint the problem.

Thanks for doing this!

ChrisOfBristol, 2020-03-29:
Images not showing in your post.
> Then you can clone the repository
Got this far and waiting for it to finish.

ChrisOfBristol, 2020-03-29:
Found diylc-swing/DIYLCStarter.launch, right-clicked then selected debug-as and DIYLC...
Program windows show. Couldn't find a "suspend button to pause the execution". Clicked the square red icon in the Console tab.
Console shows the attached. 

There were a couple of errors which look unimportant to me:

> 21:33:50,117  INFO [main] DIYLCStarter - Starting DIYLC with working directory /home/chris/git/diy-layout-creator/diylc/diylc-swing
> 
> (java:5765): Gtk-WARNING **: 21:33:50.883: Theme parsing error: gtk.css:2:18: Expected semicolon
> 
> (java:5765): Gtk-WARNING **: 21:33:50.883: Theme parsing error: gtk.css:5:31: Expected semicolon
[image]

ChrisOfBristol, 2020-03-29:
Splash screen always appears, program window 50% of attempts. If it does appear I can use the program, this is the last line in the console. 

> 21:50:06,741  INFO [SwingWorker-pool-1-thread-4] ConfigurationManager - Saving configuration

When I close DIYLC's window this is the last line in the Console:

> 21:50:21,921  INFO [AWT-EventQueue-1] ConfigurationManager - Saving configuration

bancika, 2020-03-29:
try to go to Debug perspective, it should give you access to debug section which shows where the execution is stopped

[image]

[image]

ChrisOfBristol, 2020-03-29:
I'm not seeing those tabs, so can't be finding debug perspective.
[image]
Could you give me precise directions, like Files>Save?

bancika, 2020-03-29:
Debug perspective is the bug button on to top left corner.

bancika, 2020-03-29:
Btw, I just realized that you have the debug panel opened but in your case it's on the far left :)

bancika, 2020-03-29:
So next thing you can do is press the yellow Pause button in the toolbar and see what happens with the Debug panel. If will show you where each of the threads is stopped

[image]

ChrisOfBristol, 2020-03-29:
OK, done, but the copy stack progress bar has done about 99.9% and is waiting, can I press the red button?

bancika, 2020-03-29:
Not sure what it means, haven't seen a progress bar there, but let's try :)

bancika, 2020-03-29:
Oh, and I forgot to mention. The pause button is for the selected thread only. You can pause and unpause all threads under DIYLCStarter in Debug window by selecting them and clicking the pause.

ChrisOfBristol, 2020-03-29:
I just left it on org.DIYLC...
If I click stop the details disappear. Instead I have screenshotted the whole list.
[image]
[image]
[image]
[image]

Strange that the DIYLC window sometimes appears and sometimes doesn't.

bancika, 2020-03-29:
Don't click the red stop, just the yellow pause. Is the above state what you get after a few seconds when things clear out and it hangs without anything changing anymore?

bancika, 2020-03-29:
Found some more information on what you can do outside of eclipse. If you run the app from the console and wait for it to a couple of seconds to hang, you can do this

* If the application appears to be hung and the process appears to be idle, then the first step is to try to obtain a thread dump. If the application console is available, then press Control+\ (on Oracle Solaris or Linux), or Control+Break (on Windows) to cause the HotSpot VM to print a thread dump.*

maybe it gives something useful?

ChrisOfBristol, 2020-03-29:
Yes. I wonder if the randomness of whether it works or not could be due to a race between processes? If one wins all is well, but if the other one does it hangs.

bancika, 2020-03-29:
when you say it works, you mean it starts all the way and you can draw?

ChrisOfBristol, 2020-03-29:
Yes, completely normal.
I assume console is another word for terminal, will try that.

bancika, 2020-03-29:
Thanks.  I'll analyze these and get back to you

ChrisOfBristol, 2020-03-29:
OK, thanks for your help with Eclipse.

bancika, 2020-03-30:
I read about it and put a tentative fix in...can you try again through
eclipse? You just need to pull the fix using right click on diylc-swing ->
Team -> Pull.

Cheers,
Bane

ChrisOfBristol, 2020-03-30:
I have done that and it hasn't made any difference.

bancika, 2020-03-30:
I see. Which is the most recent version that works for you? I can try to
diff the code between that version and the next one and maybe that will
help pinpoint the problem.

Thanks

ChrisOfBristol, 2020-03-30:
To quote my original post; "This used to run on Ubuntu 19.04 but won't on 19.10." Unfortunatley I didn't note which version of DIYLC that was. I'll do some tests.

bancika, 2020-03-30:
I installed the latest ubuntu with latest Java and couldn't reproduce it :(

ChrisOfBristol, 2020-03-30:
The finger points at me!

bancika, 2020-03-30:
Haha. It depends. If some older version works fine, that it ould be me.

ChrisOfBristol, 2020-03-30:
Tried 47, that does even less - no splash screen, just finishes. 

> chris@desktop:~/debian/diylc47$ ./run.sh
> 20:12:22,969 DEBUG [main] DIYLCStarter - Java version: 1.8.0_242-8u242-b08-0ubuntu3~19.10-b08 by Private Build
> 20:12:22,970 DEBUG [main] DIYLCStarter - OS: Linux 5.3.0-42-generic
> 20:12:22,970  INFO [main] DIYLCStarter - Starting DIYLC with working directory /home/chris/debian/diylc47
> 
> (java:4270): Gtk-WARNING **: 20:12:23.103: Theme parsing error: gtk.css:2:18: Expected semicolon
> 
> (java:4270): Gtk-WARNING **: 20:12:23.103: Theme parsing error: gtk.css:5:31: Expected semicolon
> 20:12:24,202 DEBUG [Thread-1] DIYLCStarter - Starting font pre-loading
> 20:12:24,202  INFO [Thread-1] DIYLCStarter - Dynamically loading font: square721_bt.ttf
> 20:12:25,416  INFO [main] ConfigurationManager - Initializing configuration
> 20:12:26,007 DEBUG [Thread-1] DIYLCStarter - Finished font pre-loading
> Exception in thread "main" java.lang.ExceptionInInitializerError
> 	at org.diylc.swing.gui.MainFrame.<init>(Unknown Source)
> 	at org.diylc.DIYLCStarter.main(Unknown Source)
> Caused by: com.thoughtworks.xstream.converters.ConversionException: Cannot construct org.diylc.core.Template as it does not have a no-args constructor : Cannot construct org.diylc.core.Template as it does not have a no-args constructor
> ---- Debugging information ----
> message             : Cannot construct org.diylc.core.Template as it does not have a no-args constructor
> cause-exception     : com.thoughtworks.xstream.converters.reflection.ObjectAccessException
> cause-message       : Cannot construct org.diylc.core.Template as it does not have a no-args constructor
> class               : org.diylc.core.Template
> required-type       : org.diylc.core.Template
> converter-type      : com.thoughtworks.xstream.converters.reflection.ReflectionConverter
> path                : /map/entry/list/org.diylc.core.Template
> class[1]            : java.util.ArrayList
> converter-type[1]   : com.thoughtworks.xstream.converters.collections.CollectionConverter
> class[2]            : java.util.HashMap
> converter-type[2]   : com.thoughtworks.xstream.converters.collections.MapConverter
> version             : null
> -------------------------------
> 	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:79)
> 	at com.thoughtworks.xstream.core.AbstractReferenceUnmarshaller.convert(AbstractReferenceUnmarshaller.java:65)a:66)
> 	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50)
> 	at com.thoughtworks.xstream.converters.collections.AbstractCollectionConverter.readItem(AbstractCollectionConverter.java:71)
> 	at
> 	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.jav com.thoughtworks.xstream.converters.collections.CollectionConverter.addCurrentElementToCollection(CollectionConverter.java:79)
> 	at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:72)
> 	at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:66)
> 	at com.thoughtworks.xstream.converters.collections.CollectionConverter.unmarshal(CollectionConverter.java:61)
> 	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
> 	at com.thoughtworks.xstream.core.AbstractReferenceUnmarshaller.convert(AbstractReferenceUnmarshaller.java:65)
> 	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
> 	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50)
> 	at com.thoughtworks.xstream.converters.collections.AbstractCollectionConverter.readItem(AbstractCollectionConverter.java:71)
> 	at com.thoughtworks.xstream.converters.collections.MapConverter.putCurrentEntryIntoMap(MapConverter.java:90)
> 	at com.thoughtworks.xstream.converters.collections.MapConverter.populateMap(MapConverter.java:78)
> 	at com.thoughtworks.xstream.converters.collections.MapConverter.populateMap(MapConverter.java:72)
> 	at com.thoughtworks.xstream.converters.collections.MapConverter.unmarshal(MapConverter.java:67)
> 	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
> 	at com.thoughtworks.xstream.core.AbstractReferenceUnmarshaller.convert(AbstractReferenceUnmarshaller.java:65)
> 	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
> 	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50)
> 	at com.thoughtworks.xstream.core.TreeUnmarshaller.start(TreeUnmarshaller.java:134)
> 	at com.thoughtworks.xstream.core.AbstractTreeMarshallingStrategy.unmarshal(AbstractTreeMarshallingStrategy.java:32)
> 	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1058)
> 	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1042)
> 	at com.thoughtworks.xstream.XStream.fromXML(XStream.java:922)
> 	at org.diylc.presenter.Presenter.<clinit>(Unknown Source)
> 	... 2 more
> Caused by: com.thoughtworks.xstream.converters.reflection.ObjectAccessException: Cannot construct org.diylc.core.Template as it does not have a no-args constructor
> 	at com.thoughtworks.xstream.converters.reflection.PureJavaReflectionProvider.newInstance(PureJavaReflectionProvider.java:71)
> 	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.instantiateNewInstance(AbstractReflectionConverter.java:428)
> 	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshal(AbstractReflectionConverter.java:233)
> 	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
> 	... 29 more
> chris@desktop:~/debian/diylc47$ 
> 
>

ChrisOfBristol, 2020-03-30:
48 gets stuck here for a while without showing splash screen
**20:26:39,260  INFO [SwingWorker-pool-1-thread-2] UpdateChecker - 11 updates found
**

at update checker without showing a splash screen.

ChrisOfBristol, 2020-03-30:
49 shows splash screen then sticks here:
```
20:31:38,053 DEBUG [SwingWorker-pool-1-thread-2] PhpFlatProxy - Connecting to: http://www.diy-fever.com/diylc/api/v1/getAnnouncements.php
20:31:38,235  INFO [SwingWorker-pool-1-thread-2] UpdateChecker - Trying to download file: http://www.diy-fever.com/update.xml
20:31:39,681  INFO [SwingWorker-pool-1-thread-2] UpdateChecker - 10 updates found
```

bancika, 2020-03-30:
Then it must be something about your machine. What about your java version? What happens when you type "java -version"?

In my installation that works fine it says

openjdk version "11.0.6" 2020-01-14
OpenJDK Runtime Environment (build 11.0.6+10-post-Ubuntu-1ubuntu119.10.1)
OpenJDK 64-Bit Server VM (build 11.0.6+10-post-Ubuntu-1ubuntu119.10.1, mixed mode, sharing)

ChrisOfBristol, 2020-03-30:
I posted that earlier, but here it is again:
I'm on this:
openjdk version "1.8.0_242"
OpenJDK Runtime Environment (build 1.8.0_242-8u242-b08-0ubuntu3~19.10-b08)
OpenJDK 64-Bit Server VM (build 25.242-b08, mixed mode)

bancika, 2020-03-30:
you did, sorry, I got lost :)

I think I used this command to install mine

sudo apt install default-jdk

ChrisOfBristol, 2020-03-30:
Have beaten you to it, I used sudo apt-get install default-jre. DIYLC now works!!!

                                                 **DIYLC now works!!!** 

Why did I have the old version? I installed Ubuntu 19.10 at the beginning of the month. I ran a batch file to install loads of software, but I've just checked that and it doesn't install Java. I assume it came with the installation of Ubuntu, but why the crappy old version???

bancika, 2020-03-30:
wow, that's great news! Glad it worked in the end :)

closed 2020-03-30 by bancika

ChrisOfBristol, 2020-03-30:
Sorry to have wasted so much of your time on this...

bancika, 2020-03-31:
No worries. At least now you are ready to start coding Java in Eclipse :)

ChrisOfBristol, 2020-04-19:
I'm up to about 14 languages now and I've had enough! Python is my current one.
I've tried that sort of software and can't be bothered with it, I just use a text editor and a terminal. No doubt it's useful for other languages or more complex programs than I write.