bancika/diy-layout-creator #483: Version 57.0 saved files cannot reopen in Windows 10

ajayrauhl, 2019-12-09:
[log-2019-12-09 10.45.10.txt](https://github.com/bancika/diy-layout-creator/files/3937692/log-2019-12-09.10.45.10.txt)

Hi, i've saved files in my windows 10 from diylc version 57.0 but i can't reopen it again. may i know why?

1) i tried saving in C drive as well but it did not help. 

and i love your software, it's really helpful since previous versions!

kindly help me on this issue as i need it for school projects.

bancika, 2020-03-23:
Can you please upload one of the files you created with DIYLC? I don't see in the log that you tried to reopen any of the files. See only save actions.

bancika, 2020-12-11:
closing due to inactivity. Please comment with more info to reactivate.

closed 2020-12-11 by bancika