bancika/diy-layout-creator #484: Mac Download reports as damaged

pslebow, 2019-12-20:
Attempt to download latest Mac OS version
[image]

MynahMarie, 2019-12-23:
I'm having the same issue... Please let me know if there's a way to fix

bancika, 2019-12-23:
It's mentioned on the first page

[image: image.png]

martinloren, 2019-12-23:
Ive latest version of MacOS and made it work with a mod I think I've send on and old post.

bancika, 2019-12-23:
How can I automate this, so my new builds are packaged like that?

pslebow, 2019-12-23:
What is mentioned on first page. This is not an issue with a non-approved app - it says the app is damaged.

bancika, 2019-12-23:
That is apple's way of reporting that an app is not approved. It's
definitely not damaged. Here's one of the links that have more detailed
explanation  https://secure.clcbio.com/helpspot/index.php?pg=kb.page&id=323

pslebow, 2019-12-23:
Ah. I see in Sierra must go to terminal to allow the “anywhere” option to appear.  Thanks!

martinloren, 2020-03-04:
This is the last version (working) for MacOS.
[DIYLC.app.zip](https://github.com/bancika/diy-layout-creator/files/4286538/DIYLC.app.zip)

bancika, 2020-03-22:
Can anyone with Mac OS give this version a try? I tried it on Mojave VM and it works fine, but it dies after starting on Sierra VM. Can't figure out the reason yet.

[diylc-4.0.1-osx.zip](https://github.com/bancika/diy-layout-creator/files/4365494/diylc-4.0.1-osx.zip)

closed 2020-03-25 by bancika

cenobitedk, 2020-04-19:
Do this
https://www.youtube.com/watch?v=6fqzb4qpgcs
or this
https://www.youtube.com/watch?v=3py0kgq95Hk