bancika/diy-layout-creator #486: schematic pot.meter symbol rotation

KENTUCKYTUKKER, 2019-12-24:
the schematic potentiometer symbol can not be rotated or mirrored. Although selection table in the transform selection menu can be opened the functions are no working. If you click "YES" for exclusion it still does not work

pixelwaster, 2019-12-24:
You can rotate the symbol before you place it by just right clicking.
After placement, select the part and -
   flip and rotate it using right click -> edit selection 
   use ctrl+E     (^E for use old timers when the net was a DoD project)
   uss edit -> edit selection in the drop down menu
   double left click to bring up the edit selection dialog box.
The last and really hairy way is to edit the *.diy project file in a text editor like Notepad++
      (been in there, done that, didn't get a T-shirt)
[image]

bancika, 2020-04-20:
fixed for 4.5.0

closed 2020-04-20 by bancika