bancika/diy-layout-creator #487: DIYLC won't open .diy files - old or new

king-custom-electonics, 2020-01-04:
I've just managed to make the latest version of DIYLC work on my Mac Book. When I try to open up a .diy file (I have hundreds) it shows an empty folder. Even if I create and save a new file the folder still shows as empty. It's not empty. Any ideas?

bancika, 2020-01-11:
can you please upload the log file? it should be in user folder under .diylc/logs

orlandwright, 2020-03-14:
Hi, I'm experiencing the same issue on Mac Catalina. I can't find directory .diylc to pull any logs.

orlandwright, 2020-03-14:
Some more info for you. Files will open correctly from the "recent files" menu option. Trying to open the same files by clicking on them results in a blank board. Trying to find them via "open" results in blank directories.

bancika, 2020-03-27:
Here's where to find the logs. Just replace "admin" with your username.

[image]

king-custom-electonics, 2020-03-27:
Thank you! I couldn't find them (I guess you knew that). It turned out to a permissions issue with OSX. I've never had this issue but suddenly DIYLC lost access to the disk drive. Remembering my IT days (mid-90s) I thought to check permissions. This is a problem you may not encounter again. My issue seems to have been a fluke.
Thanks for getting in touch!
Best,Alex

bancika, 2020-03-27:
Great!

closed 2020-03-27 by bancika