bancika/diy-layout-creator #488: Expand selection doesn't work

animalejo, 2020-01-04:
The expand selection option doesn't work

Video link: [https://www.dropbox.com/s/2c7yahg1aneztsp/IMG_1475%20%281%29.mp4?dl=0](url)

for be clear, this happen with any copper trace

bancika, 2020-03-29:
I checked the code and it is done like that on purpose. Expend selection only works for non-conductive components, so traces are not included. Closing for now.

animalejo, 2020-03-29:
but in the oldest versions, this works perfectly my friend

cheers

bancika, 2020-03-29:
Yes, I know. Older versions used a totally different logic for this
feature. But once I implemented netlists I switched it to use netlist
calculations for everything.

bancika, 2020-03-30:
I remembered the case that lead me to replace the logic...in cases like this if you expand to all connected, R1 will also get included because it overlaps with the trace and that's wrong. With netlists it doesn't happen, but I understand that not including traces at all is a different problem altogether...would need to think of a better way

[image]

animalejo, 2020-03-31:
Dude you are awesome!
well for someone's (like me) that option was so helpfull
hope you reconsiderate

Cheers!!

bancika, 2020-03-31:
I think I have a better solution. Will include it with the next version
that will go out very soon.

Cheers

closed 2020-03-31 by bancika