bancika/diy-layout-creator #492: desktop icon in the ZIP -- not really an issue

peterjosvai, 2020-01-28:
hi, 
since we can launch DIYLC 
from a "desktop launcher" (talking about Linux, but must be similar in Micro$oft environment, too), 
it is a logical step to give it an icon, too :)

which could be created from the splash.png by resizing it to 48x48, for example...
and it could be added in the next zip file...

(I attach what I've just made like that)
[image]

bancika, 2020-03-23:
Hi,
how do I create a desktop launcher for an app on Linux? Do you have experience with that?
I made a dedicated icon that could work.
Cheers

[image]

bancika, 2020-03-25:
Including a folder with the app and DIY file icons in various formats. This will be out with 4.0.1

[image]

closed 2020-03-25 by bancika