bancika/diy-layout-creator #493: when desktop launcher opens DIYLC, it will be VERSION 3.0 running instead of 3.57, mysteriously -- small "issue", but weird :)

peterjosvai, 2020-02-12:
hi, 

I am more than satisfied with DIYLC, love it, and use with zero glitches...
this is, however something weird:

I've created a desktop launcher (linux)...
in that, the path is this: 
/home/jepe/my-Apps/diylc-3.57.0/run.sh 

and the run sh looks like this from "inside":

java -splash:/home/jepe/my-Apps/diylc-3.57.0/splash.png -Xms512m -Xmx2048m -Dorg.diylc.scriptRun=true -Dfile.encoding=UTF-8 -cp /home/jepe/my-Apps/diylc-3.57.0/diylc.jar:lib org.diylc.DIYLCStarter

- - -
I have NO idea why version 3.0 is running...
but I would imagine it like the jar file -- diylc.jar -- the "DIYLCStarter" has some problem with something :) with the paths...

I couldn't view it in text editor...
- - - 

in any case, if it's not so painstaking to figure out how fix this, I'd be cool if someone could do it...

thanks anyhow!

Peter

bancika, 2020-03-20:
DIYLC version is dynamically read from update.xml file...if somehow it cannot see or access that file, it will show up as 3.0

bancika, 2020-03-24:
Hey, can you help me build the official desktop launcher for linux? I'm not too familiar with how linux packages are built.

peterjosvai, 2020-03-24:
hi, sure,
but I'm not a programmmer... or a Linux expert... only  a user...
BUT anything I can help you with, I'll gladly do :) 
but all I know is how to create a launcher...
very simple, I use Linux Mint, which is Ubuntu and Debian... (the order of adoptations is: Debian, Ubuntu, Mint :)) Debian being the original :)

so, as a text file, a launcher looks like this (saved in home/Destop)

[Desktop Entry]
Name=DIYLC 3.57
Exec=/home/jepe/my-Apps/diylc-3.57.0/run.sh     <<-- the location of the diylc  directory/run.sh
Comment=
Terminal=false
Type=Application
Icon=/home/jepe/my-Apps/icons/DIYLC.png         << obvious...

- - -

OKAY, one more thing...
linux packages work like...
(sorry if it's known to you already :) ) 

there are different "linuxes"...
the are called "distros", like distributions...
and each of them is an independent .. OS...

and they, speaking of Linux Mint, a very popular one, Ubuntu is BIIIIG,m LInux MInt is COOL, not so big, but big enough :)
so, they have software repositories... "repos"...
where they offer software for download...

these repos, as far as I know, require a "decent" developing work (regular updates, if needed), 
but any GOOD software could get a place there...

DIYLC defintely SHOULD !!!
it is perfect for any linux distro... including Ubuntu, Debian or MINT, or any other...

they only greenlight a software (an application) once they have tested it and it has proven to work well..
but DIYLC dreally works! :)

anyhow, 
I'd gladly help, and if I can, I will... but I have to mention, I'm no programmer (except for PHP)

Peter

bancika, 2020-03-25:
Thanks for the info. I thought that launcher is somewhat of a deployable
package.
I'll look what it takes to deploy to Ubuntu for starters.

Cheers

peterjosvai, 2020-03-25:
cheers!

love your work!!!
it is simply immense!
it's not only an app, but an activism, making an app for "addicts" :) :)
"addicts" of music and electronics

totally respect You!! :) and I'm not alone with that, as you know :)

Peter

bancika, 2020-03-25:
Thank you for your kind words. Stay safe!

closed 2020-03-25 by bancika