bancika/diy-layout-creator #495: Request: Docs for Linux install

TwoFistedJustice, 2020-03-02:
Is there a set of docs for installing this on Linux? 

I see the Snap died and the Flatpack evidently never went anywhere. [This tutorial](https://www.unixmen.com/install-diy-layout-creator-ubuntulinux-mint/) is out of date.

I'd really like to use DIY Layout Creator, but I just have no idea how to get it working on Linux. I don't imagine it is all that hard? But if someone could add instructions to the Readme.md or something, that would be really helpful.

bancika, 2020-03-02:
You can just unpack somewhere and use run.sh to start it.

TwoFistedJustice, 2020-03-03:
[](https://github.com/bancika/diy-layout-creator/blob/wiki/Manual.md)
The instructions here don't work for me. I have Ubuntu 18.04. 

I have openjdk-8-jre installed. But I have not idea how make your software run on my system. 

I extracted the zip file,  typed `./run.sh`
and got back `bash: ./run.sh: No such file or directory`

bancika, 2020-03-03:
It's definitely there

[image: image.png]

TwoFistedJustice, 2020-03-03:
I'd really like to use your software. But it just isn't working for me. And I do not know why. Could you please help me to get it up and running?

bancika, 2020-03-03:
Just download the latest multi-platform release from Releases section of
the website

[image: image.png]

then go to downloads and extract the archive

[image: image.png]

finally, open terminal, go to the extracted folder and type ./run.sh. That
should do the trick.

[image: image.png]

TwoFistedJustice, 2020-03-03:
Okay, thank you for your patience.

I got it to work. The problem was one of perception. I was not using the .zip archive that contained the .exe file because I perceived that as a Windows only file. I was trying to use the source code zip.

The way to get it to run in Linux:

Download:  diylc-[version number].zip

Extract the archive into an empty directory.

Open a terminal in that directory.

Set run.sh to be executable by typing

`chmod +x run.sh`

then run the program by typing:
`./run.sh`

Which is the same as

`sh run.sh`

With your permission, I will fork your repo and update Manual.md

bancika, 2020-03-03:
Absolutely, thanks!

TwoFistedJustice, 2020-03-04:
Done and sent.

closed 2020-03-04 by TwoFistedJustice

bancika, 2020-03-05:
thank you, appreciate the effort!