bancika/diy-layout-creator #496: Spice Netlist generation error

TomKamphuys, 2020-03-02:
Some component (e.g. resistor and capacitor) have 3 controlPoints. 2 controlPoints for the connections and if I'm not mistaken, the third is for the label position. In generating the netlist summary DIYLC loops over c.getControlPointCount(), which for some components contains an extra label controlPoint. This results in an erroneous netlist.

bancika, 2020-03-25:
Are you sure? I tried placing two resistors and the netlist that came out seems fine

[image]

TomKamphuys, 2020-03-25:
Well.... This is what I get:

[image]

Version 3.57

I've also looked at the code and even 'fixed/improved' it in a very hacky way in my clone.

bancika, 2020-03-25:
I see it now when I enable movable labels. OK, I'll add to the queue to fix.
Tnx

bancika, 2020-03-26:
will include in 4.0.1

closed 2020-03-26 by bancika