bancika/diy-layout-creator #501: Zoom with touchpad on Mac is broken

iiimonfiiire, 2020-03-27:
With DIYLC open, hold down command key while scrolling in/out on the touchpad. 
You will notice that the interface appears to zoom in and out randomly without any real correspondence to user interaction. 

Reproduced on: Macbook Air w/ macOS Mojave 10.14.6

bancika, 2020-04-20:
I don't think there's apple touchpad support in java jet

htor, 2020-08-16:
correct me if i'm wrong, but doesn't java support touchpad movement regardless of platform? or do you  also, it seems to me the movement gets messed up on macos because its trying to pan in x y while at same time zooming?

bancika, 2020-09-18:
I know nothing about macs really, never worked on one :)

bancika, 2020-12-11:
should be fixed. Pinch zooming works on all platforms now. Ctrl +/- as well.

closed 2020-12-11 by bancika