bancika/diy-layout-creator #502: V4.0.1 won't run on OSX with V11 JVM

chipshorter, 2020-03-27:
Before I start, I'd like to thank you for the time and effort you have put into this product!

I've never had much luck with diylc on OSX, but usually I can manage it get it to run via the run.sh shell script.

Today I installed V4.0.1, but it doesn't seem to recognise a JVM 11 as being valid. I'm running 

from running java -version...

java version "11.0.1" 2018-10-16 LTS
Java(TM) SE Runtime Environment 18.9 (build 11.0.1+13-LTS)
Java HotSpot(TM) 64-Bit Server VM 18.9 (build 11.0.1+13-LTS, mixed mode)

When I attempt to launch diilc, I get a dialog with...

"This application requires that Java 8 or later be installed on your computer. Please download and install the latest version of java..."

bancika, 2020-03-27:
does older version of DIYLC run?

chipshorter, 2020-03-27:
I'm running 3.57 via a modified run.sh script, basically I added a cd at the beginning as below ( I think that's all I did, but it was a while ago now)

cd "/applications/java applications/diylc-3.57.0"
java -splash:splash.png -Xms512m -Xmx2048m -Dorg.diylc.scriptRun=true -Dfile.encoding=UTF-8 -cp diylc.jar:lib org.diylc.DIYLCStarter

I've no doubt the above solution would also work for V4.0.1, but it seems a bit like cheating ;-)

As I remember, the MAC application complained that I didn't have a Java 1.6 installed, and this was the best workaround I could find. I've never managed to run the "native" MAC app of any version, but I'm pretty sure it was always because of it requiring an old JVM version that I didn't have.

bancika, 2020-03-27:
Yeah, that's why I upgraded to Java 8...hoping to avoid that annoyance with
macs. But it's neverending :)

closed 2020-04-04 by bancika