bancika/diy-layout-creator #503: V 4.0.1 for Win - won't boot - error message: Java Virtual Machine Launcher - Could not find the main class org.diylc

bmccowan, 2020-03-27:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

bancika, 2020-03-27:
Can you try to open the console and type

java -version

and see what you get?

bmccowan, 2020-03-28:
Thanks! It came up 1.6.0_22 So I installed the newest Java and uninstalled the old - so far looks like it works.
Stay safe!

bancika, 2020-03-28:
Cool. You too!

closed 2020-03-28 by bancika