bancika/diy-layout-creator #505: Wishlist - to import a netlist

ChrisOfBristol, 2020-03-31:
It's possible to export a netlist from DIYLC. This is part way towards importing a netlist, which would be a useful feature. Just importing the components would be good, doing anything with the netlist might be complicated and not really necessary. 

(This is mentioned, but lost among many other issues in #171.)

bancika, 2020-04-01:
well I thought about it, but it's just too complex at this point...the biggest questions are 

- how do you place those component on the layout?
- how do you know which resistors/capacitors to put? do they want to draw a schematic or a pedal layout or an amp layout?

ChrisOfBristol, 2020-04-02:
_how do you place those component on the layout?_

In a column to the right of the canvas or a row underneath it.

_how do you know which resistors/capacitors to put?_ 

The netlist file specifies component name, type and value and 2 other factors.

_do they want to draw a schematic or a pedal layout or an amp layout?_

I don't understand this - I just use DIYLC to draw stripboard layouts.

The example below shows components imported from the netlist file to the right of the canvas ready to be manually placed.

It also shows a panel which has been loaded from the imported netlist file with a list of nets that haven't yet been connected. As components are manually connected, nets could be removed from this list. The existence of the option Config>Highlight Connected Areas suggests that related functionality already exists in DIYLC.

[image]

ChrisOfBristol, 2020-04-02:
**Automation of component placement and connection.**
Opinion:In the applications that I have seen, automatic placement of components makes a mess. It's better to start by manually placing the components, before attempting to link them automatically. However even this is dubious, resulting in a rats-nest, at least some manual placement is necessary.
 ****
IMHO For the type and size of circuit that DIYLC users create, automation of component placement and connection is messy and more trouble than it's worth for the user - let alone the author of the software.

bancika, 2020-04-02:
It's relatively easy to import components while ignoring the connections
between them.

My question about component types is - say you need to import a capacitor.
Which one do you pick? Ceramic, Electrolytic, Axial/Radial Film...

ChrisOfBristol, 2020-04-02:
Here is a list from gEDA for VeroRoute;

> [
> C3
> CAP_FILM
> 82n
> 
> 
> ]
> [
> C2
> CAP_ELECTRO_200
> 1u
> 
> 
> ]
> [
> C1
> CAP_CERAMIC1
> 820p
> 
> 
> ]
It would be necessary to have a list of component types from somewhere.

bancika, 2021-01-02:
I'm toying around this idea. Do you have a list of commonly used components that could be mapped to DIYLC components? I thought that for starters it would have some predefined mappings in the code, but it can also learn through usage when user changes default behavior or maps additional types

[image]

ChrisOfBristol, 2021-01-02:
(See email too)
 QUCS component name/qucs field no of component detail/component detail/veroroute name
<Qucs Partlist V1>
'R' '' '' 'RESISTOR'
'potentiometer' '' '' 'PADS3'
'Varistors' '' '' 'RESISTOR'
'C' '13' 'neutral' 'CAP_ELECTRO_200_NP'
'C' '13' 'polar' 'CAP_ELECTRO_200'
'L' '' '' 'INDUCTOR'
'Tr' '' '' 'PADS4'
'Switch' '' '' 'SWITCH_ST2'
'Relais' '' '' 'PADS4'
'Vdc' '' '' 'PADS2'
'Vac' '' '' 'PADS2'
'LEDs' '' '' 'LED'
'LEDs_red' '' '' 'LED'
'LEDs_yellow' '' '' 'LED'
'LEDs_green' '' '' 'LED'
'LEDs_infrared' '' '' 'LED'
'_BJT' '9' 'npn' 'TO92'
'_BJT' '9' 'pnp' 'TO92'
'JFET' '' '' 'TO92'
'GND' '' '' 'PAD'
'OpAmp' '' '' 'DIP8'
'OpAmps' '' '' 'DIP8'
'Diode' '' '' 'DIODE'
'Z-Diodes' '' '' 'DIODE'
'RTD' '' '' 'DIODE'
'_MOSFET' '9' 'pfet' 'TO220'
'PMOSFETs' '' '' 'TO220'

ChrisOfBristol, 2021-01-04:
Now it's live it will be very interesting to see the feedback. I do think it will be an extremely useful function, you may need to offer different file formats, but the feedback should suggest those. On the other hand it might turn out that most people designing small to medium sized circuits for their own use have either got the schematic (as an image) from somewhere else, or just use pen and paper🙁

Using the net information in the netlist.
A function which the program uses with the nets information to automatically place the components is a far more complex problem, in fact I suspect it has a lot in common with the [Travelling Salesman Problem](https://en.wikipedia.org/wiki/Travelling_salesman_problem) I have tried VeroRoute and a long time ago when I had Windows i used a program called Stripboard Magic.

    Veroroute has a strange way of looking at stripboards - basically it tries to arrange things as you would on an etched board. This doesn't seem like a great idea to me, it means loads of solder bridges among many other snags. It bunches everything together which is a bad idea on a PCB unless it is absolutely essential. Its algorithm doesn't succeed in connecting many nets and you are left with a strange method of completing them. My trial circuit was very untidy and I gave up trying to complete it.

    Stripboard Magic did a much better job. However even a small circuit resulted in a large number of wire links, so the board would be bigger than necessary and there would be more work in wiring it. Not really good enough.

There are better approaches. To start with the user should be able to set some parameters - like closeness of soldered holes etc. Options:

    User places important components like chips which have lots of connections and sockets/controls which need to be in a certain position, power rails etc. Then press Connect and program places the rest of the components and connects them.
    Components are shown as linked together, but user places them all. Fritzing does this (from an internal schematic). It looks like a tangle to start with.
    Have a list of nets, they get ticked off as the user connects them.

All very complicated and a lot of work, and perhaps not of very great utility, so I really don't think that would be likely to become a high priority.

closed 2021-02-14 by bancika